MADEAPPS/newton-dynamics #185: Crash on Exit vs2017

TrevorCash, 2019-09-21:
[image]

I have only seen this assert happen once so far since updating to the latest sdk.  Seems like some object is not being freed as GetCount() == 1.  My NewtonPhysicsWorld Wrapper hasn't shown any object deletion issues recently.  but not sure.

JulioJerez, 2019-09-22:
some how you created something that is not be destroyed before deleting the world.

TrevorCash, 2019-09-22:
ok thats what I suspected.  I'll ll look into it

JulioJerez, 2019-11-12:
Hey TrevorCash did you get this resolved?
I committed an improvement that makes the  joint solve faster when dealing with loops, like you are doing, plus is soul be far more accurate and robust. Al least that the theory. 
I have not tested enough thought, so next time you get later code, beware of this update.

TrevorCash, 2019-11-12:
sounds great!  will multiple joints in parrallel configurations be more robust?

JulioJerez, 2019-11-12:
yes. that's the idea.
you can do a quick check wit the standard joint demo that teh SDK is set to.
if you run in debug mode you will see that the scene run a pretty much contact speed regardlesss of what happens. 

nwo if you go to file ...\sdk\dgPhysics\dgSkeletonContainer.cpp line 927
and uncomnet the line, 
//m_blockSize = 0;
	if (m_blockSize) {

you will see the phys time oscilation by a rage of more or less 10 ms,  specially when yo utry to grab a lined body, because here picj code make close loops.
depend of the machine. I addiction the solution if fat more accurate.
Now all bilateral joints on joint confirmation are always exact only the use set the option soft joint.
Contacts are still integrative, so what happen is that inaccuracy a pushed to the joints where the error is inditiguacble for the exact solution.  

No sure if you saw the vehicle Dave and I have nee working for lat few weekd, but is uses that method. This is a huge deal my friend.

JulioJerez, 2019-11-12:
one correction. whe you say
"sounds great! will multiple joints in parrallel configurations be more robust?"

if you mean give two body connected the with two joint, liek for example a hinge and a slider. 
so engine nwo do tha kind of Non Physics physics, Netwon si no on tha camp.
when I speak of loop I mean kinematic loop
here is a clean definition with a picture: http://planning.cs.uiuc.edu/node117.html

IN newton a join can no have more than 6 indeeds DOF, if is ha more is blow up, because ill condition system of equation. It can have more than 6 dof is the combination amount to a limited dif.
for example tow bodies connected by a joint when the ith the floor  the, ther are more did but the get violated by friction, motors, or you name it, the net effect is  aft eteh matrix is reduced is that it neve have more tha 6 dof.
this is still very important concept is Newton.

TrevorCash, 2019-11-14:
Ok I understand that a joint can only have 6dof.  

Does newton internally resolve multiple joints to 1-6dof?  For example here is a picture:

If I add 2 simple slider joints in parallel (lets say each slider has no twist allowed) that is still 1 dof.  Does newton internally combine the 2 joints into 1 joint?  Or does it just simulate both joints?  if it does - will the total strength be more than if it was a single joint?  (this is how it seemed to work in my game last time).  It would be kindof cool if newton did internally combine.  I can imagine cases where  50 or so sliders are all made in parallel made by the player, and optimizing this would be beneficial.

(sorry about the sideways pic:)
[image]

TrevorCash, 2019-11-14:
Thinking about it - I guess if any joint even deviates a little - the system would go into deadlock in this case.

JulioJerez, 2020-03-08:
I guess this was fixed, if not you can re open.

closed 2020-03-08 by JulioJerez