MADEAPPS/newton-dynamics #190: Compile error dModelAnimTreePoseBlender.h

uromastix87, 2019-11-05:
just pulled ran `mkdkir build`, `cd build`, `cmake ..,` `make`

```
[ 81%] Building CXX object applications/demosSandbox/CMakeFiles/demosSandbox.dir/sdkDemos/demos/AdvancedPlayerController.cpp.o
In file included from /home/rfebbo/Documents/git/newton-dynamics/applications/demosSandbox/sdkDemos/DemoCamera.cpp:17:
/home/rfebbo/Documents/git/newton-dynamics/applications/demosSandbox/sdkDemos/toolBox/toolbox_stdafx.h:168:10: fatal error: dModelAnimTreePoseBlender.h: No such file or directory
  168 | #include <dModelAnimTreePoseBlender.h
```

JulioJerez, 2019-11-05:
but that file is in the download.
I just try the CMake script in windows and is worked for me.

uromastix87, 2019-11-06:
The file is there as "dModelAnimTreeposeBlender.h" as opposed to "dModelAnimTreePoseBlender.h"

uromastix87, 2019-11-06:
same problem with:
"dAnimationRagDollEffector.h" is actually called "dAnimationRagdollEffector.h"

this one is referenced multiple ways so changing the file name didn't work for me like the last .h file.

JulioJerez, 2019-11-06:
oh I see now, case sensitive file name.
fixed. Thanks

JulioJerez, 2020-03-08:
I believe this was fixed.

closed 2020-03-08 by JulioJerez