MADEAPPS/newton-dynamics #192: Continuous collision detection not working

Mergull, 2019-11-12:
NewtonBodySetContinuousCollisionMode(body, 1) seems to not working. It doesn't change anything.
Using NewtonCollisionCollideContinue directly works properly.

JulioJerez, 2019-11-13:
yes this is correct, I need to re enabled because I was try to get the GPU solver going.

can you try  using two substep instead of 1 
when you create the engine just call  
NewtonSetNumberOfSubsteps (m_world, 2);

the continue collision  is a big problem whet many cores and GPU,  because is no possible to no longer single out island to solve individually by one core or thread and advance them to the moment of impact.  Instead many joint a put together solve at once per thread.

the engine now solve 16 joint per core once. and for GPU that number will probable 256.
after I get GPU functional I will see if running say 8 subset is sufficient to justify not doing CCD internally, is not the I will reenable CCD.

what I am doing is that the will be an immediate mode solver that will take care of all that sequential complex task and let the brute force solve do what is does best with many joints.

Mergull, 2019-11-14:
Thanks for answer.

I'm working on VR tennis simulation game. Even with 16 substeps, balls tunneling through net or racket. Currently I'm using NewtonCollisionCollideContinue function for detection racket-ball collision, but it would be very helpful if continuous collision worked out of the box.

There is currently any way to test GPU solver?

By the way what method for continuous collision detection is used by Newton? It's work super accurately, even with high rotation speed.

JulioJerez, 2019-11-14:
> I'm working on VR tennis simulation game. Even with 16 substeps, balls tunneling through net or racket.
ha I see, yes that as strong argument.
I remind me this game back is 2005 https://futurepinball.com/
is was 300fps and still need CCD fo rthe ball and the flipper.
My guess a tennis is not as fast as the flipper but is is the same problems yes.

Ok I will enable the CCD dynamics this weekend, it should no be that complicated.
few question if you can answer.

1- at what frequency you run you simulation. I hope is not 60, for a simulation grade as high speed you need something like 120 or may even higher 180 or 240 for not other reason that better integration quality. plsu since this is VR your refresh rate is 90 or 120 anyway.

2-yes the Dynamics CCD should make it very nice, in Newton dynamics CCD is not like other engine that find contact ahead of time and declare a contact. This prevent the interpenetration but is an atrocious simulation.
In newton the method actually advance the island to the moment of collision and apply impulse solve passes, is this method that generates the high quality simulation.

The CCD method is the JGK with some tweaks, by is may help in your case because the ball is sphere so is immutable to angular velocity.

Mergull, 2019-11-14:
Simulation frequency is 240. 16 substeps give good results, but not perfect.

> Ok I will enable the CCD dynamics this weekend, it should no be that complicated.

Thanks!

JulioJerez, 2019-11-17:
ok I believe CCD is enable now. 
I did not test much but I believe no much change on hat part since change the way Island are build.
please try and let me knwo if this is enough. 
I do not remember if I changed stuff in teh collision system. since I made a flat array intead of an STL style container, but does seem that part has to be revisied.

Mergull, 2019-11-18:
I've tested it, and CDD still not working for me. 
If you set body velocity to (0,30,0) in ContinueCollision.cpp (in demos), ball will fall through terrain.

JulioJerez, 2019-11-18:
oh yes, that seemed too eassy did it.
I had to change the colilsion so that contact at ccd distance are tagged valid even if the are not colliding I added a quick hack in function void dgBroadPhase::DeleteDeadContact()
plus contact buffer needed to be resized for ccd pairs and that seem to make work.

later I can made the proper change so that CCD is deem active only if the are fast or else it will cause problems with large island. 
but is good enaoug for testing if is works, please check it out again.

Mergull, 2019-11-18:
CCD works perfectly now.

JulioJerez, 2019-11-18:
oh cool.
try sync one more time I made the correct change to teh collision, so that it only declare CCD when the pair is going to collide a head of time. 

also can you make a small video so that I can see what you are working on, if possible?

Mergull, 2019-11-18:
It seems to work.

Unfortunately I can't show any video to public because of confidentiality clause. If you want I can share some information about project in private message/mail.

By the way... I'm using Unity with Newton plugin fork.

JulioJerez, 2019-11-18:
yes it should not make difference.  The last fix was jus so that bodies that a flag CCD do no issue active joint whenn they are moving away from the other body of the joint.    
Glad you are using Unity, a least I know the late build work with it.
of the video, do not worry. I was just curios. 
Anyway us know if you get any other problems and need assistance or advice.

Mergull, 2019-11-18:
Unity plugin from original repo has bug with deleting objects, and some minor issues overall (and don't compile because of deleted dCustomVehicleManager). I've fixed this issues in my fork. I want make pull request but I don't like some changes which I've made (I've found that current collision notification callback don't work correctly. It looks like callback is called only when contact joint is valid from one update to the next. So I moved callback call from loop in C# side, to C++ inside NewtonContactProcess callback. Problem is multithreading. Calling any Unity main thread bounded function (like Debug.DrawRay, Sound.Play) crashing Unity.). If I find any good solution for this problem I can make pull request with all fixes. Probably I need this solution anyway because I want to change collision properties based on this callback or even make own collision response. Maybe I can make different pull request with fixes only.

I've registered account on forum last week and I'm waiting for approval. It should be better place for asking for advice. Especially if I want to ask about something which isn't Newton bug.

JulioJerez, 2019-11-18:
oh yes 
I suspended use registration because I was only getting spams from malicious sites trying to plug links to other site. 
Mergull does not seem to be you user name, because I did no find it.
if you tell me your user name I can active your account.

Mergull, 2019-11-18:
My username is "Mergul" (additional 'l' is because Mergul already exists).

JulioJerez, 2019-11-18:
thanks, I activated your account now.

Mergull, 2019-11-18:
Thanks. Probably issue thread can be closed now.

closed 2019-11-18 by Mergull