MADEAPPS/newton-dynamics #197: Linking failed when building OpenHoW

sabas, 2020-01-28:
Hello,
I was trying to compile the [OpenHoW](https://github.com/TalonBraveInfo/OpenHoW) project under Linux but I'm stopping at the last linking step due to some missing library reference I can't seem to fix, and I was wondering if you could help me troubleshoot this.
To give more context, the developers build the project fine under Windows, while me and another user bot have a similar error. I tried to add -ldl -lpthread in some places but I hadn't any success...

```
/usr/bin/ld: ../3rdparty/newton-dynamics/lib/libdgPhysics.a(dgWorldPlugins.cpp.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libdl.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]:  [src/engine/CMakeFiles/OpenHoW.dir/build.make:1018: bin/OpenHoW] Error 1
make[1]:  [CMakeFiles/Makefile2:500: src/engine/CMakeFiles/OpenHoW.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
Thanks a lot!

sabas, 2020-01-29:
Closing, it was solved by ordering CMakeLists
https://github.com/TalonBraveInfo/OpenHoW/commit/0166463ffb5203ee1358e92102918bc2e27d1e4a

closed 2020-01-29 by sabas

JulioJerez, 2020-01-30:
Hi, glad you figure out.
Can't you tell me what you did to get it fix or provide a patch for other
people.
Thanks

sabas, 2020-01-30:
Hi @JulioJerez it's in the commit I linked, it was changed the order of the linking libraries (and added dl)

```
-    target_link_libraries(OpenHoW pthread openal SDL2)
+   target_link_libraries(OpenHoW openal SDL2 pthread dl)
```

JulioJerez, 2020-01-30:
oh you I see, you changed the order of the library on your cmake script.

I am glad you gusy are given a try to the engine, there has been many perofroem and quality improvemen on 3.14
Are you guys use the latest build from the depot head or the latest stable release?

sabas, 2020-01-30:
cc @hogsy because I'm not so much involved in the actual coding 😆

hogsy, 2020-01-30:
@JulioJerez we have a fork which has the changes from the depot head from around at least a month ago. We pretty much just pull changes across when they're needed.