MADEAPPS/newton-dynamics #208: rror: no matching function for call to 'dgInterlockedExchange'

AlexandrParkhomenko, 2020-03-14:
https://aur.archlinux.org/packages/newton-dynamics-git/

newton-dynamics/sdk/dgCore/./dgList.h:432:42: error: no matching function for call to 'dgInterlockedExchange'
        dgListNode* const first = (dgListNode*) dgInterlockedExchange ((void**) &m_first, (void*) node);
                                                ^~~~~~~~~~~~~~~~~~~~~
/p/newton-dynamics-git/src/newton-dynamics/sdk/dgCore/./dgTypes.h:541:19: note: candidate function not viable: no known conversion from 'void **' to 'dgInt32 *const' (aka 'int *const') for 1st argument
DG_INLINE dgInt32 dgInterlockedExchange(dgInt32* const ptr, dgInt32 value)

JulioJerez, 2020-03-14:
that was me tryinm to make a look free inerstion to a list, but did no relly work.
The function is not used, I see how it can be a problem with some compilers.
I commented it out,

if you sync it should be ok.
thanks for the report.

AlexandrParkhomenko, 2020-03-22:
Everything is good. Thanks.

closed 2020-03-22 by AlexandrParkhomenko