MADEAPPS/newton-dynamics #209: Memory Leak

vovannn012347, 2020-03-25:
I compiled sdk and added it as unmanaged dll for C#
Application creates around 600 bodies with two types of mesh and pairs them with ball and socket joints.
Whenever i call NewtonInvalidateCache  and then NewtonDestroy 1.5 mb of memory is leaked. This is newton specifically because i did run a test that only added bodies and destroyed newton world afterwards.
Using NewtonInvalidateCache decreases leak from ~6 to 1.5 mb
This is not an urgent issue on modern computers, but in current apllication different body conditions are tested with different variables what leads to hundreds or thousands of tests what results in a huge memory leak.

used methods:
MaterialCreateGroupID

NewtonBodyGetMass
NewtonBodyGetMatrix
NewtonBodyGetForce
NewtonBodySetMatrix
NewtonBodySetForce

DestroyAllBodies
MaterialDestroyAllGroupID
NewtonInvalidateCache
NewtonDestroy
NewtonCreate

NewtonUpdate

NewtonMaterialSetSurfaceThickness
NewtonMaterialSetDefaultSoftness
NewtonMaterialSetDefaultElasticity
NewtonMaterialSetDefaultCollidable
NewtonMaterialSetDefaultFriction

NewtonCreateConvexHull
NewtonCreateDynamicBody
NewtonBodySetCentreOfMass
NewtonBodySetMassMatrix
NewtonBodySetForceAndTorqueCallback
NewtonBodySetMaterialGroupID
NewtonBodySetCollidable

NewtonConstraintCreateBall
NewtonJointSetCollisionState
NewtonJointSetStiffness
NewtonBallSetConeLimits

JulioJerez, 2020-03-25:
are you using NewtonInvalidateCache to track memory?
that funtion does no do that, it only delete interna memory states, like contacts an initial guess velaue for iteratove solver, along with some other velue.
btw NewtonInvalidateCache is no uptodate, so if you use detrminisk results it will probably fail.
in later deveemomenpt this ha lost some of it relevat vecause fo the imposibility of getting muticore to work deterministic even on the same machine. 
some peopel clain then can make detrminsitic mutithreading, 
teh problem wi that is the to get that the working load of each job has to be balanced with extar passes. by give that thread to not get the same priorities for os to os, is even possible to get a multicore running slower that single thread. Even at teh hardware level some core are faster that others and that something can't be detected at the application level.
so the best approach I foundd is to have the workload in a queue and have each thread processing the next job. the only hint to speed this is to have teh queue prioritize jobs by the execution cost.
this way thread that are more free do more work.
Thsi seem to be teh tread that OS and hardware is following, therefore the function for supporting deterministic execution has lost importance when using multithreaded.
even when using single threaded, we still have the problem of simd units with the different width, for example a dot product in a avx can yieldd different results that a dot produce on a sse, a multadd operation in a amd using sse4.1 can yield diffrent result that mul + add on a avx. 
these are hardware barriers that render determinism impossible except even when using scalar operations. 
for these reason there function NewtonInvalidateCache no really usefull for determonism.

when you call it, the memory reduction is the deletion of the invalid joints. The memory for collisions, bodies, and intermediate buffers do not get deleted unless you delete the engine.
the engine should not leave memory leaks at all.  this is something I have taken great care.

JulioJerez, 2020-03-25:
> Side issues: ball and socket ignores twist angle, sdk misses declared hinge joints api implementation(could not find with per-file search also).

are you using build in joints? thsoe are deprecated.
The ballSocket in the dCustom joint should handle all functionality correctly.
there are test in the standard joint demo to check each joint individually.

vovannn012347, 2020-03-27:
I am sorry, this is another issue, just tested newton memory consumption and it appears to not be it.
I was wrong.
This bug report will leave here just to blame my stupidness.

closed 2020-03-27 by vovannn012347