MADEAPPS/newton-dynamics #210: Am i doing something horribly wrong? Multiple issues, hardcoded assertions, etc

george-tsiros, 2020-03-28:
compiled in vs2017 community 15.9.21 (latest as of writing)
win10 sdk 10.0.17763.0
toolset vs 2017 (v141)

downloaded source from github, master branch, opened applications\demosSandbox\projects\visualStudio_2015\demosSandbox.sln

i let it retarget it and without touching anything started a new debugging instance for demosSandBox 

preferences do not work, but as i see a dAssert(0) in there i suspect they were never meant to work, so i am not interested in that.

"using the newton mesh tool" fails an assertion:

```
demosSandBox_d.exe!dCustomBallAndSocket::SubmitAngularAxis(const dMatrix & matrix0, const dMatrix & matrix1, float timestep) Line 342
 at sdk\dcustomjoints\dcustomballandsocket.cpp(342)
demosSandBox_d.exe!dCustomBallAndSocket::SubmitConstraints(float timestep, int threadIndex) Line 417
 at sdk\dcustomjoints\dcustomballandsocket.cpp(417)
demosSandBox_d.exe!dCustomJoint::SubmitConstraints(const NewtonJoint * const me, float timestep, int threadIndex) Line 216
 at sdk\dcustomjoints\dcustomjoint.cpp(216)
demosSandBox_d.exe!NewtonUserJoint::JacobianDerivative(dgContraintDescritor & params) Line 109
 at sdk\dgnewton\newtonclass.cpp(109)
demosSandBox_d.exe!dgWorldDynamicUpdate::GetJacobianDerivatives(dgContraintDescritor & constraintParam, dgJointInfo * const jointInfo, dgConstraint * const constraint, dgLeftHandSide * const leftHandSide, dgRightHandSide * const rightHandSide, int rowCount) Line 484
 at sdk\dgphysics\dgworlddynamicupdate.cpp(484)
demosSandBox_d.exe!dgWorldDynamicUpdate::BuildJacobianMatrix(dgBodyCluster * const cluster, int threadID, float timestep) Line 240
 at sdk\dgphysics\dgworlddynamicssimplesolver.cpp(240)
demosSandBox_d.exe!dgWorldDynamicUpdate::ResolveClusterForces(dgBodyCluster * const cluster, int threadID, float timestep) Line 665
 at sdk\dgphysics\dgworlddynamicssimplesolver.cpp(665)
demosSandBox_d.exe!dgWorldDynamicUpdate::CalculateClusterReactionForcesKernel(void * const context, void * const worldContext, int threadID) Line 459
 at sdk\dgphysics\dgworlddynamicupdate.cpp(459)
demosSandBox_d.exe!dgThreadHive::QueueJob(void(*)(void * const, void * const, int) callback, void * const context0, void * const context1, const char * const functionName) Line 322
 at sdk\dgcore\dgthreadhive.cpp(322)
demosSandBox_d.exe!dgWorldDynamicUpdate::UpdateDynamics(float timestep) Line 138
 at sdk\dgphysics\dgworlddynamicupdate.cpp(138)
demosSandBox_d.exe!dgWorld::StepDynamics(float timestep) Line 942
 at sdk\dgphysics\dgworld.cpp(942)
demosSandBox_d.exe!dgWorld::RunStep() Line 1001
 at sdk\dgphysics\dgworld.cpp(1001)
demosSandBox_d.exe!dgWorld::TickCallback(int threadID) Line 1037
 at sdk\dgphysics\dgworld.cpp(1037)
demosSandBox_d.exe!dgAsyncThread::Execute(int threadID) Line 120
 at sdk\dgcore\dgmutexthread.cpp(120)
demosSandBox_d.exe!dgThread::dgThreadSystemCallback(void * threadData) Line 201
 at sdk\dgcore\dgthread.cpp(201)
demosSandBox_d.exe!invoke_thread_procedure(unsigned int(__stdcall*)(void *) procedure, void * const context) Line 92
 at minkernel\crts\ucrt\src\appcrt\startup\thread.cpp(92)
 [ somewhere here some external code was called but i deleted the message so i don't know if it is here or above or whatever]
demosSandBox_d.exe!thread_start<unsigned int (__stdcall*)(void *)>(void * const parameter) Line 115
 at minkernel\crts\ucrt\src\appcrt\startup\thread.cpp(115)
```

matrix0 is:

```
-       m_front {...}   dVector

        m_x 1.00000000  float
        m_y 0.000000000 float
        m_z 0.000000000 float
        m_w 0.000000000 float
-       m_up    {...}   dVector

        m_x 0.000000000 float
        m_y 1.00000000  float
        m_z 0.000000000 float
        m_w 0.000000000 float
-       m_right {...}   dVector

        m_x 0.000000000 float
        m_y 0.000000000 float
        m_z 1.00000000  float
[... 237 lines cut ...]

... in sdk\dVehicle\dMultiBodyVehicle\dMultiBodyVehicleEngine.cpp line 151

```
static int xxxx;
void dMultiBodyVehicleEngine::UpdateAutomaticGearBox (dFloat timestep) {
    m_metricInfo.m_gearsCount = 4;
    //  m_gearTimer--;
    //  if (m_gearTimer < 0) {

    dFloat omega = dAbs (m_omega);
    xxxx++;
    if (xxxx > 200) {
        SetGear(dEngineInfo::dGearRatioIndex(dEngineInfo::m_firstGear));
        return;
    } else {
        SetGear(dEngineInfo::dGearRatioIndex(dEngineInfo::m_neutralGear));
        return;
    }

```

is someone trashing the source tree? what is this?

Am I missing something? Did something happen recently?

JulioJerez, 2020-03-28:
oh wow, that's a long list of problem, I check it out and you are right. I made a mistake in teh ball socket joint, last week and I have no look at teh engine since. It is fix now. 
I verify that these bug are fixed.
"using the newton mesh tool"  works
"gyroscopic precession" works
"primitive collision"  this is a very old demo, that sow how the engine can be used as  collsion only library
"simple box stacks" this is not meat to be run in debug, in fact you are getting faster fps than I in debug. in release you should get under 10 ms per frame, but if you run multithreaded and async update the physics update sould be almost free.
"compound collision shape", "non-uniform scaled collision shape" fixed
"dTractorModel::MakeHydraulic" fixed
"passive ragdoll" should work.
"dMultiBodyVehicleEngine" this one Is temporarily disabled because I am merging the player controller 
"single body car" is a contribution by Dave Gravel, should be working.
"basic player controller" yes there was a problem, is fixed now

"simple cloth patch" this is on hold until the secudn half of the year when get the GPU solver ready
for now I need to get the full body IK an dthe GPU solver. 

please you sync, then you can check again.

JulioJerez, 2020-03-28:
thanks for the detailed report

george-tsiros, 2020-03-28:
Neat! That was fast... thank you very much for your work! I will keep testing.

Sadly i am not very good at c++ so i can only help a little.

I would like to use this under .net with c#, i see sdk\dWrappers\csharp\ is that work in progress for a c# wrapper?

again, thank you for your work and your efforts!

closed 2020-03-28 by george-tsiros

JulioJerez, 2020-03-28:
That's the wrapper for the unity plugin.
It uses swiig for making a manage csharp wrapper.

george-tsiros, 2020-03-28:
it *looks* like gpu acceleration works. Tried 32bit debug, 32bit release and 64bit release. When i switch to gpu solver and restart the demo, I see that after letting the scene settle, it is about twice faster (5ms instead of 10ms, vega 56, stock) and if i move the vehicles blindly forward it is a bit faster.

are you a physicist? When the system is analyzed, does it make any attempt to simplify the degrees of freedom or does it apply all constraints as they were initially described? 

I see that in the msvc/visualstudio projects, the compilation options *to me* look unoptimized. 1) you usually use gcc/other so you haven't tweaked them 2) you let us choose what we want 3) you have checked that it doesn't make a difference 4) something else? 🤔

I have many more questions, but for now I will let you do your work, the fixes you made are enough for me to start reading the source more closely 

again, thank you very much for your work and your time

JulioJerez, 2020-03-28:
no, I am not a physicist, my degree is in mechanical Engineering.
It engien is implement a holonomic reduced coordinate system, therefore each joint is represented by at most 6 degree of freedon, so there is not need simplify any further.
I does have to simplify the degree of freedomm for contact joints. and I have tried several methods over the years.   
1-desposing joint into Eigen values and selection the rows with the must dominant eigen values.
this is very elegant solution, teh code is still in teh engine but disabled, but is has the problem that it may deem important contact as redundant, so it is better to use geometrical methods. 
I am now using a hybrid method for reduce dof for contacts. Basically a pre reduction using geometric projection over the convex hull formed by the contact points. them the  linear system solver clamps contacts that reach max values, making in effect similar to singular value decomposition. It seems to work quite well.

the compiler option are not really that important, it only matter the one for a particual instruction set. Since the engine is open source, any one can chose the compiler options they want to try.

george-tsiros, 2020-03-29:
WAIT.
you've been a coder since 97!?!