MADEAPPS/newton-dynamics #213: Make use of dynamic linker optional

JayFoxRox, 2020-05-01:
I'm statically linking Newton, and to my surprise it will still `LoadLibrary`, `GetProcAddress` and `FreeLibrary` at runtime:

https://github.com/MADEAPPS/newton-dynamics/blob/master/sdk/dgPhysics/dgWorldPlugins.cpp#L146-L150

The goal of static linking (in my particular case) is to avoid any use of the dynamic linker, because the target platform, original Xbox (while being win32), does not actually have a dynamic-linker (although I might eventually implement one myself). But also for other platforms, I expect that not everyone wants (or requires) plugins.

For portability, it would be nice if WorldPlugins could be made optional in the CMake configuration.

JulioJerez, 2020-05-02:
right again, I no added preprocessor  DG_USE_PLUGINS for optionally adding plugins.
the hand made buidl have the on by default but teh cmake script can configure them optimally.

JayFoxRox, 2020-05-03:
I did not try this switch yet, but I reviewed the commit and noticed that the linux headers will still be included, even if `DG_USE_PLUGINS` is defined; so for linux-like platforms, people would still have to provide dummy header files, even if they don't have a dynamic linker (and despite the headers being unused).

JulioJerez, 2020-05-03:
The linus header are needed for any posit type system ther are not conditional on teh plugins.
as far as I knwo they are under teh __Linux__ prepprocessor.

#if __linux__
#include <dlfcn.h>
#include <dirent.h>
#endif

JayFoxRox, 2020-05-03:
There might be POSIX subsets which do not have a dynamic linker, and they will not even provide `dlfcn.h`, so compilation would fail, even if `DG_USE_PLUGINS` isn't defined.

But if `DG_USE_PLUGINS` is not defined, then `dlopen`, `dlsym`, .. etc. are never actually used. The same applies to `opendir`: the code which contains them is disabled because `DG_USE_PLUGINS` is not defined.
So including `<dlfcn.h>` and `<dirent.h>` is not necessary. It's only necessary when plugins are enabled.

If you include them regardless (even if they are unused) then they must still exist. This just forces people to create an empty `dlfcn.h` and empty `dirent.h` to compile newton-dynamics - this shouldn't be necessary and could be easily avoided.

JulioJerez, 2020-05-04:
I do not know what that would be a problem, but ok, I added this

#if defined(DG_USE_PLUGINS) && defined(__linux__)
#include <dlfcn.h>
#include <dirent.h>
#endif