MADEAPPS/newton-dynamics #216: Most demos in demosSandbox do not work as intended

JayFoxRox, 2020-05-01:
Many of the demos in the sandbox do not work for me. I get crashes with some, and visual errors or lack of physics in others. Depending on the order I launch them in, I also get slightly different behaviour.
Most of the demos seem to run slow, even at high FPS (implying rendering issues of some sort).

I have used Newton around 2005, so I know it's good. But I assume that many people would be scared by this. It also has me worried wether these are bugs in the samples, or wether the solvers / core features are currently broken.

I personally assume these bugs have to do with random code being disabled to work on new solvers or other things within newton (as this is a pattern I had seen with newton in 2019 already - and also it's why I abandoned it then: lack of stable repository / very bad git history with typos and unclear changes).

If issues with the samples are not known / specific to me, then I'll go into detail as to what I consider unexpected behaviour. It will be a lot of work to document them, so I'd rather want to know if problems are expected (so they don't have to be reported).

If these issues are to be expected, then I'd still like to see it documented that the demos might be broken right now + I'd express my desire for better development practices, where "master" (or at least the last tagged release) actually works, and where the history is "clean" and descriptive.

JulioJerez, 2020-05-02:
as far as I know most all demo works. 
what demos are you having problems with?

JulioJerez, 2020-05-02:
are you using lastest stable release, if so the try getting 3.14c
some o fthe issue you see has been fixed. 

on the graphic I do not know what to say, the graphics in newton are just for showing the physics, the may run fat or slow for system to system.  
as I undertand they seem reasonably fast in most system I seen.

Hurleyworks, 2020-05-02:
Hi Julio,

I just downloaded the latest release and the Object Placement demo and the Multi geometry collision both crash here. All others are working.

JayFoxRox, 2020-05-03:
*(In hindsight, "most" is probably an overstatement, but it's still "many")*

All tests done on x64 with avx or avx_d plugin on Arch Linux using Newton v3.14c compiled locally using `cmake -DCMAKE_BUILD_TYPE=Debug`, followed by `make`.
Many of the cases where an `assert` triggers in my debug build, it will actually SIGSEGV in release builds, or demos will misbehave a lot.

When reading this, please keep in mind that people run the sandbox to get a feel for newton-dynamics capabilities, without looking at the code.

## Starting the sandbox

I'm used to just reach the compiled binary by tab completion, from my build folder:
```
mkdir build
cd build
cmake ..
make
./applications/demosSandbox/demosSandbox 
```

However, if that is done, then newton-dynamics won't find the AVX library on my machine.

Also some sample loading a file will just crash without any useful information for the user.
An example for a crashing sample when run from the build directory is "Archimedes Buoyancy":

```
demosSandbox: newton-dynamics/applications/demosSandbox/sdkDemos/toolBox/TargaToOpenGl.cpp:151: GLuint LoadTexture(const char*): Assertion `(sDepth == 3) || (sDepth == 4)' failed.
Aborted (core dumped)
```

Changing the current directory to the sandbox folder will fix this.
Ideally it would also work otherwise, or it should at least handle this more gracefully.

## All samples

None of the options in the "File" menu work for me (or I don't understand them?), except "Exit".

The help menu is always empty.

The movement of the camera feels very slow. I instinctively held Shift to go faster, but it just made it slower.

The rotation of the camera is even worse, and rotating the camera by 180 Degree using mouse-drags is virtually impossible.
I'm using a trackpoint and can use the Blender Walk/Fly-mode just fine. Also ego-shooters work fine.
But dragging across the sample window at 1366x768 over the full display width will turn the camera only about 20 degree.
In the vertical axis I need to drag 9 times across the entire window height to rotate the camera from horizontal to looking straight down. So it's about 10 degree per turn.

I also think the vertical axis is inverted for my personal preference. I think this should be configurable or swapped.

Hotkeys like F1 for resetting the sample are not advertised; F1 is only advertised by individual sample(s), but it always seems to work.

## Specific samples

### Using the newton mesh tool

It's unclear what the newton mesh tool even is.
With this being the first item on the list, the user might already be confused.
There should be a description for users who do not look at the code.

### Newton Cradle

Only has a single ball, so this is just one ball swinging around after being hit.

The ball being shot at it also slowly floats away.
As a new user might not know what to expect, this looks like not even gravity is working properly.
Because the user doesn't expect one of the balls being shot into the cradle, the user might also assume that the physics simulation simply "exploded".

### Kinematic bodies

The moving plate on the right does not have collision, but is barely ghosted.
On my first run I assumed that this was a bug in newton-dynamics.

Later I realized that it was ghosted and probably only shows collisions.
[... 109 lines cut ...]
If this is primarily about the code, then this should be displayed in the demo.

### Simple cloth patch

Crashes immediately:

```
demosSandbox: newton-dynamics/sdk/dgPhysics/dgCollisionLumpedMassParticles.cpp:254: virtual dgMatrix dgCollisionLumpedMassParticles::CalculateInertiaAndCenterOfMass(const dgMatrix&, const dgVector&, const dgMatrix&) const: Assertion `0' failed.
Aborted (core dumped)
```

When I ran this without debug mode, the cloth patch would behave like a solid piece - it did not act like cloth at all. It was like a rigid plate that also halfway intersected with ground.

### Simple soft body

Crashes immediately (after the window hangs about 3 seconds):

```
demosSandbox: newton-dynamics/sdk/dgNewton/Newton.cpp:7654: void NewtonCreateTetrahedraLinearBlendSkinWeightsChannel(const NewtonMesh*, NewtonMesh*): Assertion `0' failed.
Aborted (core dumped)
```

---

Other issues might still exist, but these are the most critical ones to me.

JulioJerez, 2020-05-03:
Six axis manipulator is a inverdynamic demostration for peopel who do robitic, they kknwo what it mean. adding object to pick add nothing to the demo, since the object of to appling the controll. 

Passive Ragdoll is a diffrent kinbg of rag doll when the joint have friction. it soeul work perfect in all cases.
Vehicles I was try to do some refeitmen to the demos. The vihobel model works I was jsu tryt to fien tune the controls.

Advanced player controller it simpel show teh cpasuel pleye cotrontll plaion a animation. The engien is no an animation system, 
but you are right the demo sould be more polished, it si just that I do not have  mosh tiem to spend in polishing demos. 
I it my hope to show enought of the physocs that is suffient for a end use to detect what he look for. 
I can't not afford, and I never will affordt, to have a team of develeper doing polished demos to show off the engine. 

mos teh demo you see in teh SDK are of tow kind. 
1 and experiment thing that I try to do of my own bolistion.
2 a repro demo to shwo a user how to solve some specif problem

JayFoxRox, 2020-05-03:
> Six axis manipulator is a inverdynamic demostration for peopel who do robitic, they kknwo what it mean.

I know that it's an Kinematic/IK demo. That portion works fine.

However, until recently, there was a `TestSpline` function which drew (what appears to be) unrelated splines to the UI (which is problematic as it's just confusing)
Recently, this was changed to draw a white spline in 3D space. However, that spline *still* does absolutely nothing for me. It doesn't seem to interact with the robot (or its code) in any way.
Looking at the code, the variable `m_splineCurve` in the robot class is unused.

The code also says the robot is supposed to track it - but this does not actually happen for me (or I don't get it): 
- If this is a broken / incomplete feature, then it should not be displayed (as it will confuse users who look at the sample).
- If this is a working feature, there should be a description as to what it's supposed to show.

> adding object to pick add nothing to the demo, since the object of to appling the controll.

An object to pick could help because the robot could actually track the object, as well as showing grip behaviour.

I believe the tracking portion is a goal with the spline (keeping the gripper on a spline), but it's not working.

> Passive Ragdoll is a diffrent kinbg of rag doll when the joint have friction. it soeul work perfect in all cases.

Here are the ragdolls how they rest after dropping (they are way too stiff and bounce like rigid objects, often landing in their T pose; in some instances knees will move after the have been on the ground for a while):

[image]

In other situations they spawn within the terrain or fall into it; the terrain height should be taken into consideration when spawning them.
You can still see their stiffness / T Pose:

[image]

So even if the joints should have friction, then the sample must be tuned to be much less stiff.
However, I think this is actually a bug, because the cloth sample is affected in a similar way (being 100% stiff):

[image]

It's acting like a rigid plane and it will intersect the floor (this is how it rests after lifting it and letting it fall down again).

> Advanced player controller it simpel show teh cpasuel pleye cotrontll plaion a animation. The engien is no an animation system,

It should use the player model / capsule from the basic player controller sample then. Having an animated character implies the animations should work. Especially with this being after "Animated player controller" in the list (which sets expectations).

>  I it my hope to show enought of the physocs that is suffient for a end use to detect what he look for.

My impression is that newton-dynamics is a good piece of software, but I can't blame anyone who will not use it because:

- The master git revision, and stable releases, often include dirty code or commits which break features. This makes it very unlikely for me to consider it for any serious project, because I can't even rely on stable versions being fully functional. This shows in the demos.
- Many of the demos assert or crash, so newton-dynamics does not feel stable enough. If the sample code crashes at startup, then users will have to assume that many other portions will also randomly crash, leak memory, or otherwise.
- Many of the demos assert or crash during use, so newton-dynamics does not seem capable of running more than a minute or two without crash. So I can't blame anyone who would question how it could support any serious application (especially with most popular applications using it being from a decade ago or more).

again: I *know* that newton has a ton of good features; but I also understand why some people would be scared away.
The project makes a really bad first impression. And the bugs persisting in stable releases will also scare developers during development (as they don't know which revision of newton-dynamics can be trusted for the release of their software).

JulioJerez, 2020-05-03:
Well if the demos are so bad, there are other physics engines that perhaps
sue better yous expectations.
I am not judging anyone who decided against it, the engine is what it is,
sorry it is not what you expected.

JayFoxRox, 2020-05-05:
> Well if the demos are so bad, there are other physics engines that perhaps sue better yous expectations.

I have been using newton-dynamics since 2005 or so (and have also used ODE and Bullet since then). I believe that newton is the most suitable solution for my current project.

However, my current project is a toolchain / SDK - so it will be used by other developers who are new to newton-dynamics. This means I need material to show that newton-dynamics works.

The demos shouldn't crash, and they should work for everyone. Currently they don't. Right now, newton-dynamics hides its capabilities behind broken demos.
Just run the newton cradle demo, and you'll see what I mean: it's confusing. It's not a newton cradle, and just looks buggy.

> I am not judging anyone who decided against it, the engine is what it is, sorry it is not what you expected.

It's not an issue with the physics portion in newton-dynamics, but just with the default configuration or hacky changes in most demos. And I consider that to be easily fixable bugs. Most of the items on my list above are like that.

For example: That cradle demo is broken because of poor development practices: [It was modified for a WIP solver change](https://github.com/MADEAPPS/newton-dynamics/commit/bee7f896ba5666fda4da9783f437eae2b4b12e10#diff-4d326e29e7e891e0edb0850f2fb1ffa1R114), but then the demo was left in this broken state.
Such experimental changes to the demo should never have been public in my opinion.

I'm willing to fix at least some of those bugs; but I would like these issues to be acknowledged first (do they only happen for me? why do they happen?). It's very hard to tell if some bugs exist for known reasons and I don't want to waste my time debugging known problems.

hypernewbie, 2020-12-08:
I second that my experience is that the demos are completely broken, and hence has lead me to this thread for confirmation.
That's ok, good news is that the physics engine works!

dlannan, 2022-12-27:
The demos are still in a bad state. The main problem I have with this, is as JayFoxRox mention, Im not sure what works, and what doesnt. For example the Simple Vehicle (which I believe is the MultiBodyVehicle.cpp?) just crashes as mentioned above, and theres no clear reason why. Then in the code, there are a number of Assert(0); as well as many #if 0 without any commenting or explanation?
Ive spent some time trying to get Newton working across a couple of platforms, and its taken a large amount of time, and very little of it (newton physics itself) is actually reliable. 
This is disappointing, because it used to be a really solid physics engine (back around 2012 ish when I used it in some demos). Just be warned, if you want to use this for vehicle physics or in game physics, it is very unstable and I would probably recommend trying something else.

JulioJerez, 2022-12-27:
I am not sure what you are referring too.
as of teh last commit yesterday all demos in 4.01 are working.
The engine builds on all supported platforms: Linux, Windows, Mac, Android. There are even automatic unit tests 
that buidl the engine on Windows and Linux each time a chnge is commited.

I just try teh vehicle and seems to work fine,

Hurleyworks, 2022-12-27:
None of the demos crash here on Windows. But the "heavy vehicle"  demo's
user interface doesn't seem to work.
None of the driving keyboard controls seem to do anything.

Also the newton_tests don't work out of the box. Here's the error message.

[image: newton_tests_error.jpg]

JulioJerez, 2022-12-28:
ah, thanks for that note:
I recent try to code the Vehicle with a steering and a Joystick.
The control work best with steering, because some one is trying a custom steering and I can not say more. 
But I realized that is very difficult to emulate analog control with keyboard, so I try the xbox Controller.
xBox controller is better, but still so anywhere what you can get with an actual steering, it is a huge difference. 

anyway you are correct, the UI controls are no working with keyboard, I will fix that.

Hurleyworks, 2022-12-28:
>
> Also the newton_tests don't work out of the box. Here's the error message.

Hmm, they are working now after I closed vs2022 and then opened it again.

JulioJerez, 2022-12-28:
hi Hurley, I just try the demo_test in windows. and I get these results.

> [----------] 1 test from StaticBody
> [ RUN      ] StaticBody.mesh
> [       OK ] StaticBody.mesh (47 ms)
> [----------] 1 test from StaticBody (47 ms total)
> 
> [----------] 1 test from RigidBodyInstances
> [ RUN      ] RigidBodyInstances.Instancing
> [       OK ] RigidBodyInstances.Instancing (310 ms)
> [----------] 1 test from RigidBodyInstances (311 ms total)
> 
> [----------] 1 test from RayCast
> [ RUN      ] RayCast.rayCastFilter
> [       OK ] RayCast.rayCastFilter (7 ms)
> [----------] 1 test from RayCast (8 ms total)
> 
> [----------] 1 test from RigidBodyNotify
> [ RUN      ] RigidBodyNotify.MoveWithUnitForceDifferentMasses
> [       OK ] RigidBodyNotify.MoveWithUnitForceDifferentMasses (4 ms)
> [----------] 1 test from RigidBodyNotify (4 ms total)
> 
> [----------] Global test environment tear-down
> [==========] 8 tests from 7 test suites ran. (409 ms total)
> [  PASSED  ] 8 tests.

are you referring to different tests?

Hurleyworks, 2022-12-28:
The tests all work now for me too. Originally I got that error message I
showed you but after quitting vs2022 and restarting the tests are  working
fine now.

dlannan, 2022-12-28:
I have been using 3.14 across multiple platforms, because the version 4 I have had all sorts of problems compiling on Linux, Android, and Windows. Note: This is building static libs and embedded code, not using shared libs. With version 3.14 I was able to get built across Windows, Android and Linux however _all_ of the demos (as mentioned above previously from @JayFoxRox) failed to work fully. Please read his detailed information - it is still in this state. 

I am referring to 3.14, and multi platform. Linux, OSX, IOS, Android and Win. Generally earlier versions did tend to match. I know maintaining 3.14 and 4 is a big job, If version 3.14 isnt going to be supported further (as it appears) then maybe  putting a readme.md in the repo declaring this, or maybe the state of the samples in 3.14 are not current (or something like that).  I cant move my older projects to 4 so its unlikely I can continue using it.

The reality is, if I need to make major upgrades, then its much more likely the projects I use will probably migrate to a newer engine like Jolt or similar. Again this isnt a reflection on Newton itself - it is a good physics system.

JulioJerez, 2022-12-28:
Well, I am sorry for your inconvenience with 3.14
I do not remember making changes that render tge engine un usable. I know
for a fact, there are people still using 3.14 in multiple platform without
those errors, but again they are not using it for the demos.

The corrent version of newton is 4.02
There will be not more time dedicated to any older version.
You do not have to kept trying using something that is causing you so much
aggravation.
Try using something else and consider your time a leaning experience.
Good luck.

JulioJerez, 2022-12-28:
ok Hurleyworks
I fixed the GameController input. 
there multiple bugs.
The first was that the mapping was originally a one to one using the keyboard.
but when using an xbox or a wheel controller, I realized that it needs a virtual mapping because each controller is different
so I made a virtual mapping that maps 32 keys to a vector. must the scan code from keyboard fell outside that range.

I did the mapping for the key use to drive the vehicle, but the moving parts for the heavy vehicle will only work with Game controller.
I will have to maps the key, 1, 2, 3, 4, .... to some key in file ndGameControllerInputs so for now they will only work when using a game controller. 
but all vehicle should be drivable with keyboard or game controllers

anyway thanks for that warning.

Hurleyworks, 2022-12-29:
I tried the latest version and the heavy vehicles are now drivable with the
keyboard. But there is no sound like there is in the basic vehicle demo

JulioJerez, 2022-12-29:
Ah yes, the heavy vehicle demos does not have sound.
When I copy from the basic vehicle there was not sound support. So the demo
base diverged.

Thank for testing, it

closed 2026-01-16 by JulioJerez