MADEAPPS/newton-dynamics #218: Linking with C possible but blocked by trivial things

hecktest, 2020-05-05:
Newton's API works with C and C-compatible languages, but requires extra header manipulation at the moment.
Newton.h seems to have been prepared with C in mind, but it includes dgTypes.h which is a straight up C++ header. Most of the definitions and includes from that header are not necessary for binding to the plugin's API, we only need the numerical typedefs. The rest should probably live in another header.

Another problem is that the empty structs declared in place of classes are not the standard way to declare an opaque type in C, and many compilers will reject them. The correct way to do that is:
`typedef struct NewtonMesh NewtonMesh;`

JulioJerez, 2020-05-06:
yes that is correct dgTypes.h was added to get the export type, by that goes again teh c interface
I added this define below and that remove teh depency on  dgTypes.h
#define DG_LIBRARY_EXPORT __declspec(dllexport)
	#define DG_LIBRARY_IMPORT __declspec(dllimport)
	#define DG_LIBRARY_STATIC

The other c declaration is just one way of making struct declarations, the way we do it is also legal both is c and cpp, I hope this works for you.

hecktest, 2020-05-06:
Thanks for looking at this.
Actually, empty structs aren't legal in ISO C. Some compilers allow it, but others will refuse to compile it altogether.  Pelles C, for instance, rejects it.

https://stackoverflow.com/questions/755305/empty-structure-in-c

Since that portion is only ever compiled as C, it would be a good idea to just delete the braces. No compiler is going to miss them :)

JulioJerez, 2020-09-14:
oh thsi late, but maybe it si still useful I removed the braces, teh struct are now define liek this

```
	typedef struct NewtonMesh NewtonMesh;
	typedef struct NewtonBody NewtonBody;
	typedef struct NewtonWorld NewtonWorld;
	typedef struct NewtonJoint NewtonJoint;
	typedef struct NewtonMaterial NewtonMaterial;
	typedef struct NewtonCollision NewtonCollision;
	typedef struct NewtonDeformableMeshSegment NewtonDeformableMeshSegment;
	typedef struct NewtonFracturedCompoundMeshPart NewtonFracturedCompoundMeshPart;

```

also I start 4.00
it does not has C interface yet, that will come later.   
teh reason for 4.00 is to clean up lot of legacy accumulated for over r 15 years. 
also the design has become a bottleneck for new features so its time to stream line it and make it more modern.

closed 2020-09-14 by JulioJerez