MADEAPPS/newton-dynamics #223: Forum account is still not activated

wot-emul, 2020-11-27:
Hello.
I have registered on the forum and I am waiting for the activation of my account. Unfortunately, the account has not yet been activated. I tried to register again, but this time there was nothing. I have a few questions about using the physics engine. Is it possible to activate an account? The first one has the nickname "void", if I remember correctly. The second has "baly". Because I don't need two accounts, please delete the second account. Thank you.

JulioJerez, 2020-11-28:
sorry about that, I had to disable auto registration because of the malicious span.
I activated the "void" account. 
please see if you can log in.

wot-emul, 2020-11-28:
Hello.
It works. Thank you.

closed 2020-11-28 by wot-emul