MADEAPPS/newton-dynamics #227: Add cmake target aliases

jonesmz, 2021-02-27:
Currently in my project, I have to do:

```
target_link_libraries(myexe PRIVATE
    ${CMAKE_THREAD_LIBS_INIT}
    dNewton dScene dModel dVehicle
    )
```

I would much rather do this

```
target_link_libraries(myexe PRIVATE
    ${CMAKE_THREAD_LIBS_INIT}
    Newton::Newton Newton::Scene Newton::Model Newton::Vehicle
    )
```

Without the d prefix

JulioJerez, 2021-03-01:
I do not understand why is that  problem, that's just the class name. 
d or not d is just your preference, I can assure you ther will be someone who will like a different preference.

jonesmz, 2021-03-01:
I think there is a miscommunication.

By using cmake aliases, it keeps the cmake scripts easier for projects to use.

It doesn't change the semantics of the projects being able to pick which variant of the thing they are pulling in.

But the cmake scripts for newton are very hard to understand, so I did not even realize there were different versions of these libs.