MADEAPPS/newton-dynamics #228: Why are newton 3.14 and newton 4.0 subfolders?

jonesmz, 2021-02-27:
Shouldn't these be branches of the git repository?

JulioJerez, 2021-03-01:
maybe, but what is the difference?

tombish, 2021-03-01:
For me a bigger issue is the size of the current repository. It sits at over 500MB and takes some considerable time to download. Moving 3.14 into its own branch would help. Also perhaps you could split example code/binary resources into their own repository.

jonesmz, 2021-03-01:
Another difference is that if someone is tracking, specifically, 3.14 or 4.0, they wouldn't want to see differences for the other codebase in their diff viewer.

Keeping them as separate branches helps keep things simple for the person using the code.

100espressos, 2021-11-08:
Perhaps this issue can be closed/continued in a similar one, with a possible workaround posted.
https://github.com/MADEAPPS/newton-dynamics/issues/254#issuecomment-963393724

jonesmz, 2022-05-07:
I would still prefer to see newton remove the 3.14 folder from the branches of the repository that contain the 4.0 code.

The whole point of git's branch capabilities is to support having different series of the code in different branches. Making them folders in the same branch is very much not useful.