MADEAPPS/newton-dynamics #229: Use C++ namespaces instead of "nd" prefix

jonesmz, 2021-02-27:
Why are functions and classes, such as https://github.com/MADEAPPS/newton-dynamics/blob/master/newton-4.00/sdk/dNewton/ndBodyParticleSet.h being prefixed with "nd". That's what namespaces are for?

JulioJerez, 2021-03-01:
yes, but name spaces are a suggestive thing. some people like them and soem other people prefer the prefix, I am in teh later group.

Hedede, 2022-03-28:
I haven't actually seen a physics engine that doesn't do this. Havok uses the `hk` prefix, Bullet uses `bt`, PhysX uses `Np` etc

jonesmz, 2022-03-28:
And everyone else jumping off of a bridge means we should as well?

It's trivial to macro-ize this to allow for the consumer of the library to use a prefix or a namespace, at their choice.
```
#ifdef USE_CPP_NAMESPACE
#define NEWTON_NAMESPACE(symbol) nd::#symbol
#else
#define NEWTON_NAMESPACE(symbol) nd#symbol
#endif
```

But inflecting global symbols on the consumer of the library is not polite.

Hedede, 2022-03-29:
I don't see it as a big deal if everything is correctly prefixed. It doesn't really "pollute" anything. Name conflicts are unlikely, and it won't show up in the code completion unless you type the prefix.

I think it is better to keep prefixes than to "macro-ize" everything.

JulioJerez, 2022-03-30:
Yeah, of all of the reasons to reject a library, the name prefix is the
least important that I can think up.
What people has to realized is that these projects ha been going for almost
20 years.
At the time not all platforms suppted cpp, an nane spaces was a new thing
to cpp back them.
Trying to retroactively apply this kind of changes is a real big job that
does not justify the effort.
Maybe for newton 5 if we ever made one.