MADEAPPS/newton-dynamics #230: Add github actions continuous integration for sandbox demo apps

jonesmz, 2021-02-27:


JulioJerez, 2021-03-01:
I am no sure what the mean, the cmake scrip let you select the demos as an option.

jonesmz, 2021-03-01:
By adding continuous integration actions to the repository, you can set github to automatically build any pull request that someone creates, and tell you if the pull request causes your sandbox demo applications to break on any of the platforms that github supports (Linux, windows, Mac, currently).

This is useful for contributors to newton because they might not have access to all three platforms, and would be able to get feedback right away if their code causes problems.

This is also helpful because it can serve as a "known working" example of how to build and run the example applications.