MADEAPPS/newton-dynamics #231: Linker errors related to `dVector` in version 4

n0bra1n3r, 2021-03-20:
Hello guys, thank you for this amazing project!

I've just started integrating this into my game engine to replace AMD's FEMFX. I'm trying to use the latest and greatest version 4.0, but I'm getting linker errors like `error LNK2001: unresolved external symbol "public: static class dVector dVector::m_triplexMask" (?m_triplexMask@dVector@@2V1@A)` when I try to link the built library against my project. What can I do to fix it?

n0bra1n3r, 2021-03-20:
For now I've built it as a static library, which works.

JulioJerez, 2021-03-23:
that is strange all those constant has the D_CORE_API whi add dll or lib export functionality
this you add these define to your project?
_D_CORE_DLL
_D_NEWTON_DLL
_D_COLLISION_DLL

n0bra1n3r, 2021-06-01:
Thanks @JulioJerez, and sorry for replying so late. Solved my issue for now by just linking statically, but I think the issue was that I was missing a `#define` somewhere.

closed 2021-06-01 by n0bra1n3r