MADEAPPS/newton-dynamics #232: NewtonConstraintCreateUpVector equivalent in version 4?

n0bra1n3r, 2021-06-01:
Hi again, wasn't sure where to post this. I've just joined the forums but haven't been approved yet (handle is rb3), so I can't post there yet. Is there an equivalent of `NewtonConstraintCreateUpVector` in version 4.0? Trying to find a way to keep the engine from applying torque on my rigid body to keep it from rotating in any direction. I want to be able to control rotation purely by setting the matrix of the body, but keep gravity and collision under the engine's control.

Also would you consider setting up a discord or gitter channel? Would be cool to be able to have a quick chat with the experts. Unless there is already a chatroom other than the forums?

JulioJerez, 2021-06-01:
Hi, I just approved your account. 
thanks for registering.

I just added the Up Vector and Plane 2d joints. 
I have not test them yet I will do it later.

n0bra1n3r, 2021-06-02:
That's amazing, thank you @JulioJerez! Still curious about what you think about a discord channel :)

closed 2021-06-02 by n0bra1n3r