MADEAPPS/newton-dynamics #233: Linux: Attempts to symbolically link media directory with itself.

kaiwulf, 2021-06-02:
I'm attempting to compile v3.14 on Linux, I get a linking error. It's attempting to create a symbolic link with itself.

`Linking CXX executable demosSandbox
symbolic link resources folder from /home/path/to/newton-dynamics/newton-3.14/applications/media => /home/path/to/newton-dynamics/newton-3.14/applications/media
failed to create symbolic link '/home/path/to/newton-dynamics/newton-3.14/applications/media' because existing path cannot be removed: Is a directory
make[2]: *** [applications/demosSandbox/CMakeFiles/demosSandbox.dir/build.make:1025: applications/demosSandbox/demosSandbox] Error 1
make[2]: *** Deleting file 'applications/demosSandbox/demosSandbox'
make[1]: *** [CMakeFiles/Makefile2:905: applications/demosSandbox/CMakeFiles/demosSandbox.dir/all] Error 2
make: *** [Makefile:136: all] Error 2`

JulioJerez, 2021-06-02:
Try 4.00

JulioJerez, 2021-06-02:
please try 4.00
4.00 is being designed to be more user friendly than 3.14 
it already has almost all the functionality.

kaiwulf, 2021-06-02:
I was able to install using the build instructions from [here](https://newton-dynamics.readthedocs.io/en/latest/installation.html).

`cd newton-dynamics
mkdir build && cd build
cmake -DNEWTON_DEMOS_SANDBOX=ON .. && make`

That should go in the README so it's more clear how to build from source.

cmake doesn't work for 4.00. I was planning on making another Issue for that.

JulioJerez, 2021-06-02:
It should work if you just use the cmake gui.
I never used the command line.

StuAtGit, 2022-09-27:
For the record, I ran into this ticket, searching for an answer to the same issue occurring on 4.00.

```
[100%] Linking CXX executable ndSandbox
symbolic link resources folder from [REDACTED_PATH]/newton-4.00/applications/media => [REDACTED_PATH]/newton-4.00/applications/media
failed to create symbolic link '[REDACTED_PATH]/newton-4.00/applications/media' because existing path cannot be removed: Is a directory
make[2]: *** [applications/ndSandbox/CMakeFiles/ndSandbox.dir/build.make:1210: applications/ndSandbox/ndSandbox] Error 1
make[2]: *** Deleting file 'applications/ndSandbox/ndSandbox'
make[1]: *** [CMakeFiles/Makefile2:501: applications/ndSandbox/CMakeFiles/ndSandbox.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

This is using cmake on linux as well. Not sure why it's trying to link directory itself. Several other cmake projects are working fine (ogre, my own project, bullet physics, etc), so it's probably a configuration issue in Newton.

JulioJerez, 2022-09-27:
I think it is creation the make file in tge same source directory.
I set so that cmake create a make file, a project or a solution in a
separate folder.

Try that in cmake gui.
Also why don't yo use newton4 which is much better?