MADEAPPS/newton-dynamics #234: ndSandbox basic vehicle spins out of control, basic convex fracture perf

geekrelief, 2021-06-03:
I just found out about newton-dynamics today!  I checked out newton-4.00's demos and for the most part the performance looks amazing. I tried out the sandbox and ran into a couple of issues.

In the basic vehicle demo.  Twice I was able to make the car spin out of control.  I was able to capture it on video the second time.  Is this expected behavior?  I picked up the vehicle, dropped it and it started spinning uncontrollably.  https://youtu.be/yovCqnfwdIQ?t=140

In "basic convex fracture", the performance is single digits fps whereas all the other demos run between 1400-3000+ fps. 

https://user-images.githubusercontent.com/27040/120586836-1f02b700-c3e9-11eb-8e4a-fba0d171351c.mp4

Also, I registered on the forum, can you please add me?  I'm considering newton-dynamics to integrate it with Godot. Thanks!

JulioJerez, 2021-06-03:
Oh thank for looking at the engine.
On the vehicle.
At this time this is expected because the vehicle is 100%. physically
simulated with multi bodies. Not cheat of any kind.
The final production controller will fix that alone with other issues.
Basically this is fixed by adding an angular drag that will suckup some of
the kinetic angular momentum. But for now we are still trying to push the
limits of how far we get without any assistance.
Another issue is a follow camera that can make it easier to drive in the
road, right now this is hard.
Finally the roads of the track will probably be replaced with one that is
wider and fewer road side block, but thus will be after we get the second
heavy vehicle demo completed.

On the destruction.
Yes that one will be slower with a single thread.
We are trying to get a gpu open cl, or later different graphic app that
support gpu for general purpose. For these kind of massive special effects.
Things like physics particle for fluid, soft bodies and debris, will be the
target for that gpu solver.

Of course they will also work in cpu with multicore. But in a smaller scale.

Finally if you tell me your user name in the forum, I will activate your
account.

Again, thank for looking at the sdk.
Julio.

JulioJerez, 2021-06-03:
ok, 
I entered geekrelief  as user name and I activated that account.

geekrelief, 2021-06-03:
Thanks for activating my forum account. 🙏

Thanks for the explanation about how the vehicle is setup.  I'm glad you're working on an eventual fix, but I just wanted to be sure this was documented somewhere. As for all the other issues related to the vehicle demo those are all true. A small suggestion would be to allow the max velocity to be high enough to complete the loop.  The first time I got the spinning behavior was when I attempted to complete the loop and fell and hit one of the side barriers.

As for the destruction, I tried it with multiple threads, set to 11 and it it made no difference.  I'm still getting 1-2fps  I'm running a Ryzen 5 3600.

Anyway, I'll head over to the forum to continue the discussion.

JulioJerez, 2021-06-03:
did you try the account?
on the car, the vehicle has a wide range of configuration,   
one of the problems I found is that the track, is no really suitable for that kind of stunt.  
In the pass  when we where conditioning the tire contact but issuing a ray cast to project them over the ground, the car was able to make that loop.  But now that it is all strictly rigid body simulated (not ray cast) , the bumps on the road are too large for a tire to overcome, it really needs to smoother road.   

like I said it need some features, that may be one on them.  but I am still adding generic functionality. 
if you look at the code, you can see that it has a descriptor that allow to configure almost every thing that define a vehicle.  there you can make as physically plausible or as arcady as you want.  Believe me, a game designer will have a ball playing with that .

on the destruction, do no put too much emphasis on the performance of that demo. this was an experiment to see how far the solver can be stretch, and it is clear that it no going to do it.  

the newton solver is designed for more correctness and accuracy that for performance, and that brute fore arrangement of joint will bring to its kneed, there are others ways to achieve that kind of effect. 
but I am leaving there until I get back to it.  

some people do not like when I do that, and they judge the engine by those incomplete demos that are work in progress or experiments , but at this point I do not care, if that is what the look for, them there are other physics solution available they can use. 

I look forward to see you in the forum.
Julio.

geekrelief, 2021-06-03:
Yes I just posted on the forum.  http://newtondynamics.com/forum/viewtopic.php?f=9&t=9728 Thanks again.

Thanks for the pointers. I will continue exploring the engine and demos. I'm still very early on in my evaluation, and I'm not assuming anything about what newton is or isn't capable of.  I just want to document things and to try to understand what's going on.  The other fracture demos set me up with an expectation performance.  It was unclear whether the demos are designed to show case ability, performance, or push things to the breaking point.  Now I understand it's all 3.