MADEAPPS/newton-dynamics #235: Linux: Newton 4.0

100espressos, 2021-06-07:
Thanks for Newton!
When a lot of good work is going on, some dust is expected. :)
For a couple days I am stuck here:

```
-D CMAKE_BUILD_TYPE:STRING=Debug -D CMAKE_INSTALL_PREFIX:STRING=out -D NEWTON_DOUBLE_PRECISION:BOOL=1
```
[ 57%] Building CXX object sdk/CMakeFiles/ndNewton.dir/dNewton/ndDynamicsUpdateAvx2.cpp.o
newton/newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:56:33: error: no member named 'm_type' in 'dBigVector'
                : m_type(_mm256_set_m128(high.m_type, low.m_type))
                                         ~~~~ ^
newton/newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:56:45: error: no member named 'm_type' in 'dBigVector'
                : m_type(_mm256_set_m128(high.m_type, low.m_type))
                                                      ~~~ ^
```
With only single precision, I get
```
[ 57%] Building CXX object sdk/CMakeFiles/ndNewton.dir/dNewton/ndDynamicsUpdateAvx2.cpp.o
newton/newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:41:11: error: always_inline function '_mm256_set1_ps' requires target feature 'avx', but would be inlined into function 'ndAvxFloat' that is compiled without support for 'avx'
                :m_type(_mm256_set1_ps(val))
                        ^
```

There may be more. This is probably a trivial fix too, but I am still getting up to speed with Newton, and look forward to contributing where I can.

Setting up CI (in Actions) should help catch many errors automatically. https://github.com/MADEAPPS/newton-dynamics/issues/230#issuecomment-788284323

JulioJerez, 2021-06-07:
oh, I see. you are try to build with double precision. 
I have not written the math classes to support that yet.   
It is not that complicated. just that no one has requested.  I can do it this weekend if you try   
double is actually very important in newton few engine user do requires that for their project, but they have no made the transition yet. 

for now can you try building without option -D NEWTON_DOUBLE_PRECISION:BOOL=1 
them I will post here when you can try with double.

100espressos, 2021-06-07:
Thanks for the fast reply!
Though I anticipate using double eventually (a nice feature!), I have no existing need, so no hurry on my account.

With single precision (no double option), I am getting this:
```
[ 57%] Building CXX object sdk/CMakeFiles/ndNewton.dir/dNewton/ndDynamicsUpdateAvx2.cpp.o
newton/newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:41:11: error: always_inline function '_mm256_set1_ps' requires target feature 'avx', but would be inlined into function 'ndAvxFloat' that is compiled without support for 'avx'
                :m_type(_mm256_set1_ps(val))
                        ^
newton/newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:41:11: error: AVX vector return of type '__m256' (vector of 8 'float' values) without 'avx' enabled changes the ABI
```

JulioJerez, 2021-06-07:
ah ok,  please sync again and 
In this file ../newton-dynamics\newton-4.00\sdk\CMakeLists.txt 

line 75, you will fine this 

	if(MSVC)
		 set_source_files_properties(dNewton/ndDynamicsUpdateAvx2.cpp PROPERTIES COMPILE_FLAGS " /arch:AVX2 " )
	endif(MSVC)

	if(UNIX)
		#set_source_files_properties(dNewton/ndDynamicsUpdateAvx2.cpp PROPERTIES COMPILE_FLAGS " /arch:AVX2 " )
	endif(MSVC)

in the UNIX statement need to be uncomments and replace the option for compiling with  avx2 in the compiler you are using.  since I do no build linux, I do not know but if you get it to work, can you tell me so that I can get it fix.

JulioJerez, 2021-06-07:
I just quickly look it up and I thing for GCC is this 
set_source_files_properties(dNewton/ndDynamicsUpdateAvx2.cpp PROPERTIES COMPILE_FLAGS " /march=haswell " )

you do not need to enable avx, because if you do they the library will assume avx and you will not be able to call it from code that is not 100% avx.

please sync and try again.

100espressos, 2021-06-07:
A different small issue I have been bypassing: Newton-4 may need some `cmake_uninstall.cmake.in` to complete.

You are correct and the current result is much better.
```
newton-4.00/sdk/dCore/dTree.h:752:11: error: use of undeclared identifier 'm_allocator'
    dAssert (m_allocator);
```

JulioJerez, 2021-06-07:
fixed, please sync again.

100espressos, 2021-06-07:
`newton-4.00/sdk/dCore/dContainersAlloc.h:75:32: error: ‘dMemory’ has not been declared`

JulioJerez, 2021-06-07:
ok fixed try again please.

100espressos, 2021-06-07:
Okay. One more here:
```
newton-4.00/sdk/dCore/dTypes.cpp:31: error: ‘__rdtsc’ was not declared in this scope
newton-4.00/sdk/dCore/dTypes.cpp: In function ‘dUnsigned64 dGetCpuClock()’:
newton-4.00/sdk/dCore/dTypes.cpp:31:16: error: ‘__rdtsc’ was not declared in this scope
   31 |         return __rdtsc();
```

JulioJerez, 2021-06-08:
fixed, try again.

100espressos, 2021-06-08:
```newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:45:35: error: ‘__m256’ does not name a type``` (and a few errors with undeclared 'm_type' presumably related)

JulioJerez, 2021-06-08:
I though we have that one fixed with this change

```
	if(UNIX)
		set_source_files_properties(dNewton/ndDynamicsUpdateAvx2.cpp PROPERTIES COMPILE_FLAGS " /march=haswell " )
	endif(UNIX)
```

JulioJerez, 2021-06-08:
does teh intrinsic type __m256 is defined different in GCC?

JulioJerez, 2021-06-08:
oh I think I knwo why,  in the linux the intrics headers are not included
```
#if (defined (_WIN_32_VER) || defined (_WIN_64_VER))
	#include <intrin.h>
	#include <emmintrin.h> 
	#include <pmmintrin.h>
	#ifdef D_USE_VECTOR_AVX
		#include <immintrin.h>
	#endif
#endif
```

I now replaced with the general  #include <immintrin.h>

please sync and try again, hopefully builds this time.

100espressos, 2021-06-08:
Progress!
```
/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/include/avxintrin.h:1318: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_set1_ps(float)’: target specific option mismatch
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/include/immintrin.h:43,
                 from newton-4.00/sdk/dCore/dTypes.h:75,
                 from newton-4.00/sdk/dCore/dCoreStdafx.h:25,
                 from newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:22:
/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/include/avxintrin.h: In constructor ‘ndAvxFloat::ndAvxFloat(dFloat32)’:
/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/include/avxintrin.h:1318:1: error: inlining failed in call to ‘always_inline’ ‘__m256 _mm256_set1_ps(float)’: target specific option mismatch
 1318 | _mm256_set1_ps (float __A)
      | ^~~~~~~~~~~~~~
newton-4.00/sdk/dNewton/ndDynamicsUpdateAvx2.cpp:41:39: note: called from here
   41 |                 :m_type(_mm256_set1_ps(val))
      |                         ~~~~~~~~~~~~~~^~~~~
```

JulioJerez, 2021-06-08:
it is still the same problem not compiling those avx2 intrinsic. 
the only think I can think off is that the option is set wrong, teh gcc doc say to use a dash instead of a forward slash
I change from this
set_source_files_properties(dNewton/ndDynamicsUpdateAvx2.cpp PROPERTIES COMPILE_FLAGS " /march=haswell " )

to this
set_source_files_properties(dNewton/ndDynamicsUpdateAvx2.cpp PROPERTIES COMPILE_FLAGS " -march=haswell " )

please try again. see if that works.

100espressos, 2021-06-08:
Correct and thanks again. Next minor..

```
newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp: In member function ‘void ndDynamicsUpdateOpencl::CopyBodyData()’:
newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:684:33: error: ‘union cl_float4’ has no member named ‘x’
  684 |                 data.m_rotation.x = body->m_rotation.m_x;
```

100espressos, 2021-06-08:
```
[ 64%] Building CXX object sdk/CMakeFiles/ndNewton.dir/dNewton/ndDynamicsUpdateSoa.cpp.o
newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp: In member function ‘void dOpenclBuffer<T>::ReadData(cl_command_queue)’:
newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:86:50: warning: there are no arguments to ‘GetCount’ that depend on a template parameter, so a declaration of ‘GetCount’ must be available [-fpermissive]
   86 |                         CL_FALSE, 0, sizeof(T) * GetCount(), source,
      |                                                  ^~~~~~~~
newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp: In member function ‘void ndDynamicsUpdateOpencl::CopyBodyData()’:
newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:684:33: error: ‘union cl_float4’ has no member named ‘x’
  684 |                 data.m_rotation.x = body->m_rotation.m_x;
      |                                 ^
```

JulioJerez, 2021-06-08:
ah, for that one in cmake ther are tow define that let boteh avx2 and opencl to be #ifdef out 

option("NEWTON_ENABLE_AVX" "enable AVX float instructions" OFF)
option("NEWTON_ENABLE_GPU_SOLVER" "enable gpu solver" ON)

of the two the avx2 is functional and  teh fastest solver so far. 
the opencl is still in development so there is not point is installing sdk and all that stuff. 
you should call cmake with   NEWTON_ENABLE_GPU_SOLVER = OFF

I am having secund thought about Opencl since is has been advandone but almost every GPU maker, 
when it come the time to address it again, I will see if we go with Vulkan or some similar. but for now it is off

if you sync again, I set OpenCL solve to off by default
please try again.

100espressos, 2021-06-08:
```
newton-4.00/sdk/dNewton/ndWorld.cpp:454:2: error: use of undeclared identifier '_mkdir'; did you mean 'rmdir'?
        _mkdir(assetPath);
        ^~~~~~
/usr/include/unistd.h:847:12: note: 'rmdir' declared here
extern int rmdir (const char *__path) __THROW __nonnull ((1));
           ^
```

JulioJerez, 2021-06-08:
ok changed to this
#if defined (_WIN_32_VER) || defined (_WIN_64_VER)
	_mkdir(assetPath);
#else
	mkdir(assetPath);
#endif
no sure if will compile, or nee a header. 
maybe later I will use teh standard file system calls but for nwo try that again.

100espressos, 2021-06-08:
It feels very close now. I am happy if you like to change focus, I can try a PR for this and look at helping with other issues myself.

JulioJerez, 2021-06-08:
did you getting compiling completely?

100espressos, 2021-06-08:
The library compiles completely and I will send a PR.
Errors in the demos could be fixed already by @iSLC https://github.com/MADEAPPS/newton-dynamics/pull/226

JulioJerez, 2021-06-08:
oh I see, you send two pull request, I will review them tonight.

JulioJerez, 2021-06-09:
ok I applied the tow PR, one when without problem, the older one had conflicts, so I apply the changes manually.
if you sync now all the changes should be there, but the patch still do not merge since is conflict even more.
but the code should have every thing that is in the patch with exception of the allocator in uppercase.  

finally I am curios to know the compiled code run.

100espressos, 2021-06-09:
Good to see iSLC's PR made it in. Update of remaining issues here in applications/

```
newton-4.00/applications/ndSandbox/toolBox/ndConvexFractureModel_0.h:47:2: error: unknown type name 'ndDemoDebrisRootEntity'
newton-4.00/applications/ndSandbox/demos/ndBasicFracture_0.cpp:76:3: error: cannot initialize object parameter of type 'ndConvexFracture' with an expression of type 'ndConvexFractureBox'
newton-4.00/applications/ndSandbox/demos/ndBasicFracture_0.cpp:96:2: error: cannot initialize object parameter of type 'ndConvexFracture' with an expression of type 'ndConvexFractureBox'
newton-4.00/applications/ndSandbox/toolBox/ndConvexFractureModel_1.h:68:3: error: unknown type name 'ndDemoDebrisRootEntity'; did you mean 'ndDemoDebrisEntity'?
```

JulioJerez, 2021-06-09:
fixed, try again.

JulioJerez, 2021-06-09:
should be defined in ../newton-4.00\applications\ndSandbox\ndSandboxStdafx.h line 120
but I refactored, try again.

100espressos, 2021-06-09:
Compile output: https://hastebin.com/aqarocoqen.typescript

JulioJerez, 2021-06-09:
ok fixed, try again

100espressos, 2021-06-09:
`Built target ndNewton
Built target ndSandbox` :fireworks: 

Thanks for your time!
Can start exploring Newton and identifying areas in which to contribute.

For future reference if one does setup github CI, there is the [guide](https://docs.github.com/en/actions/learn-github-actions) and a working example from Urho3D https://github.com/urho3d/Urho3D/tree/master/.github/workflows

Regards

JulioJerez, 2021-06-09:
did you get any of the demos running?

100espressos, 2021-06-09:
Yes; a glance, the demo seems to work as expected.

JulioJerez, 2021-06-09:
oh that's awesome.  
Happy driving

100espressos, 2021-06-10:
One offtopic: I too have a forum registration with similar username. Thanks in advance.

JulioJerez, 2021-06-10:
what is the user name so that I can activated it.

100espressos, 2021-06-10:
`espresso`

JulioJerez, 2021-06-10:
ok, is activated.

closed 2021-06-10 by 100espressos