MADEAPPS/newton-dynamics #240: Compilation warning on header guard dAnimationTwoWayBlend.h

execomrt, 2021-08-02:
newton-3.14/sdk/dAnimation/dAnimationTwoWayBlend.h:12:9: warning: '__D_ANIMATION_TWO_WAY_h__' is used as a header guard here, followed by #define of a different macro [-Wheader-guard]
newton-3.14/sdk/dAnimation/dAnimationTwoWayBlend.h:13:9: note: '__D_ANIMATION_EFFECTOR_BLEND_TWO_WAY_h__' is defined here; did you mean '__D_ANIMATION_TWO_WAY_h__'?

JulioJerez, 2021-08-03:
yes, I made a mistake when doing teh animation for 4.00
it was fixed.
thank for the warning.

has you try using 4.0 ?
it almost has all the functionality of 3.14 and is far more robust.

execomrt, 2021-08-06:
Hello, I've branched 3.14 and also resolved some compilation fixes for Android NDK. Check dgTypes.h for example. Android might not have ARM supported and can have SSE for example. I didn't test yet 4.0.

JulioJerez, 2021-08-06:
> Hello, I've branched 3.14 and also resolved some compilation fixes for Android NDK.

what chnage did you make?

JulioJerez, 2021-08-06:
also try using 4.0
some good things are coming, that you might like.