MADEAPPS/newton-dynamics #246: The player capsule doesn't collide with the chassis.

BenjaminGu, 2021-08-09:
[image]

The player capsule collides with tires.
However, it doesn't collide with the chassis.
Why is this happening?

BenjaminGu, 2021-08-09:
[image]

And player capsule doesn't collide with a scene created by BuildCompoundScene().

JulioJerez, 2021-08-09:
I don't know, my guess is that the chassis is a compound shape and the
player capsule post contact process code has a bug there. It should
collide, that's a bug that need fixing.

Thanks for the report.
I will debug it.

JulioJerez, 2021-08-09:
the bug was here
```
			//if (contact->m_isIntersetionTestOnly)
			if (contactSolver.m_intersectionTestOnly)
			{
				ndBodyTriggerVolume* const trigger = body1->GetAsBodyTriggerVolume();
				if (trigger)
				{
					body1->GetAsBodyTriggerVolume()->OnTriggerExit(body0, m_timestep);
				}
				//contact->m_isIntersetionTestOnly = 0;
				contact->m_isIntersetionTestOnly = 1;
			}

```
I was testing the flag on the contact joint in order to report collision test on the same joint, when is should test the flag on the contact solver.

ok fixed. please check again 
and thanks for the report.

JulioJerez, 2021-08-09:
It liek that you mixing thing up. I never try to see how teh player react wit teh vehicle. :)

BenjaminGu, 2021-08-09:
It works. Thank you for your immediate help.

closed 2021-08-09 by BenjaminGu