MADEAPPS/newton-dynamics #249: Generating mesh collider is very slow

SamCZ, 2021-08-12:
Hi, I have created mesh collider for my voxel mesh with around 12k vertices and it takes 24ms at minumum when generating.

This is my code:
```
dMatrix chunkMatrix(dGetIdentityMatrix());
chunkMatrix.m_posit = dVector((float)data.ChunkPos.x * 16.0f, (float)data.ChunkPos.y * 16.0f, (float)data.ChunkPos.z * 16.0f, 0.0f);

if(auto* specialVoxelMesh = dynamic_cast<SpecialVoxelMesh*>(mesh.get()))
{
	dPolygonSoupBuilder meshBuilder;
	meshBuilder.Begin();
	dVector face[3];

	const auto& indexBuffer = specialVoxelMesh->GetIndexBuffer(0);
	VertexBuffer<SpecialVoxelMeshVertexBufferEntry>* voxelBuffer = specialVoxelMesh->GetVertexBuffer(0);

	for (int i = 0; i < indexBuffer.size() / 3; ++i)
	{
		auto i0 = indexBuffer[i * 3 + 0];
		auto i1 = indexBuffer[i * 3 + 1];
		auto i2 = indexBuffer[i * 3 + 2];

		const SpecialVoxelMeshVertexBufferEntry& v0 = voxelBuffer->Get(i0);
		const SpecialVoxelMeshVertexBufferEntry& v1 = voxelBuffer->Get(i1);
		const SpecialVoxelMeshVertexBufferEntry& v2 = voxelBuffer->Get(i2);

		face[0].m_x = v0.Position.x;
		face[0].m_y = v0.Position.y;
		face[0].m_z = v0.Position.z;

		face[1].m_x = v1.Position.x;
		face[1].m_y = v1.Position.y;
		face[1].m_z = v1.Position.z;

		face[2].m_x = v2.Position.x;
		face[2].m_y = v2.Position.y;
		face[2].m_z = v2.Position.z;

		meshBuilder.AddFace(&face[0].m_x, sizeof(dVector), 3, 0);
	}

	meshBuilder.End(false);

	ndShapeInstance staticMeshInstance(new ndShapeStatic_bvh(meshBuilder));

	subObject->PhysicsBody = new ndBodyDynamic();
	//body->SetNotifyCallback(new ndDemoEntityNotify(scene, entity));
	subObject->PhysicsBody->SetMatrix(chunkMatrix);
	subObject->PhysicsBody->SetCollisionShape(staticMeshInstance);
	AuroraEngine::Physics->AddBody(subObject->PhysicsBody);
}
```

The lag happens when calling `meshBuilder.End(false);`.

My suspicion is that I'm doing something wrong, I could not find many examples on mesh colliders, could you please help me ?
Thanks.

JulioJerez, 2021-08-12:
12k vertices is a very small vertex load.
My guess is that there are many overlapping faces.

If I understand correctly, the vowel mesh is made of regular size cubes.
When two of more cubes are adjacent this will make two or more quad to face
each other.

This is very bad for the collision mesh generator, because it assume the
input mesh is a clean manifold, this means each edge can only be shared by
one or two faces.
Whe the mesh generator detect edge with more than rwo faces, it try to
separate the mesh separate set. But it does not expect to be the rule, it
expect that to be the exception, like a bug in the mesh. This is because
separating non manifold meshes into a set of manifold meshes is a
nonpolynomial time complexity algorithm.

The are few things you could do.

1- write the algorithm in a way that those opposing face are skipped. That
should not be too difficult.

2-since this is a vowel, you probably want to general the collision in real
time. For special effects like destruction or that kind of thing.

The is in the engine a file for generation similar king of meshes, marching
cube, that I was going to use for that, but I found that I would have to
put lot of time on visualization, so in end up just doing the static height
field.

This is what I can do.
I can add A new demo the generate a real time mesh, that is an infinite
grid.

Them there you can see what methos has to be whiten and you can use that to
implement you vowel call.
Basically you subclass fro the static mesh and you find those function that
are call by the contact  solver
Does that sound reasonably.?

JulioJerez, 2021-08-12:
ok If you sync, yo uwill find a new demo called ndStaticProceduralMeshCollision.cpp
this is going to create a procedural mesh liek you are tryin to do.

right now does it simple make a full polygonal bvh mesh, which is similar to what you are doing, 
but later today I will make the base class so that the application fill the overlapping faces at run time.

this is in the engine already if you look at class class ndShapeHeightfield: public ndShapeStaticMesh
but need to be formalize with the proper class, an dthe functions that need to be populated by the application

please tell me if this is what you need of I am getting it all wrong.

JulioJerez, 2021-08-13:
ok now ther is a procedural collsion mesh demo.
it implements a regular grid.  and the class look liek this

```
class ndRegularProceduralGrid : public ndShapeStaticProceduralMesh
{
	public:
	ndRegularProceduralGrid(dFloat32 gridSize, dFloat32 sizex, dFloat32 sizey, dFloat32 sizez, const dVector& planeEquation)
		:ndShapeStaticProceduralMesh(sizex, sizey, sizez)
		,m_planeEquation(planeEquation)
		,m_gridSize(gridSize)
		,m_invGridSize(dFloat32 (1.0f)/ m_gridSize)
	{
	}

	virtual dFloat32 RayCast(ndRayCastNotify&, const dVector& localP0, const dVector& localP1, dFloat32, const ndBody* const, ndContactPoint& contactOut) const
	{
		dVector segment(dVector::m_triplexMask & (localP1 - localP0));
		dFloat32 den = m_planeEquation.DotProduct(segment).GetScalar();
		dFloat32 num = m_planeEquation.DotProduct(localP0).GetScalar() + m_planeEquation.m_w;
		dFloat32 t = -num / den;
		contactOut.m_point = localP0 + segment.Scale(t);
		contactOut.m_normal = m_planeEquation;
		return dClamp (t, dFloat32 (0.0f), dFloat32 (1.2f));
	}

	virtual void GetCollidingFaces(const dVector& minBox, const dVector& maxBox, dArray<dVector>& vertex, dArray<dInt32>& faceList, dArray<dInt32>& faceMaterial, dArray<dInt32>& indexListList) const
	{
		// generate the point cloud
		dVector p0(minBox.Scale(m_invGridSize).Floor());
		dVector p1(maxBox.Scale(m_invGridSize).Floor() + dVector::m_one);
		dVector origin(p0.Scale(m_gridSize) & dVector::m_triplexMask);
		dInt32 count_x = dInt32(p1.m_x - p0.m_x);
		dInt32 count_z = dInt32(p1.m_z - p0.m_z);

		origin.m_y = 0.0f;
		for (dInt32 iz = 0; iz <= count_z; iz++)
		{
			dVector point(origin);
			for (dInt32 ix = 0; ix <= count_x; ix++)
			{
				vertex.PushBack(point);
				point.m_x += m_gridSize;
			}
			origin.m_z += m_gridSize;
		}

		// generate the face array
		const dInt32 stride = count_x + 1;
		for (dInt32 iz = 0; iz < count_z; iz++)
		{
			for (dInt32 ix = 0; ix < count_x; ix++)
			{
				faceList.PushBack(4);
				indexListList.PushBack((iz + 0) * stride + ix + 0);
				indexListList.PushBack((iz + 1) * stride + ix + 0);
				indexListList.PushBack((iz + 1) * stride + ix + 1);
				indexListList.PushBack((iz + 0) * stride + ix + 1);

				faceMaterial.PushBack(0);
			}
		}
	}

	dVector m_planeEquation;
	dFloat32 m_gridSize;
	dFloat32 m_invGridSize;
};

```

all you need to do is to make the vertex list index list mesh of the part of you procedural mesh (the voxel map)
that intersect the the aabb passed to function GetCollidingFaces
I hope that fixe your problem. 

finally one thing to remember, try to eliminate the interion faces. these are the faces that are shared by two adjacent boxes.
I am sure teh graphics algorithm remove these for visualization, so it should not be hard make do that. 
thsi make the collsion mesh a much higher quality since is can generate edge share info.

later some day, I will probably make a demo using the marching cube to show a 3d version.

SamCZ, 2021-08-17:
Hi, sorry for no response, I've not been here for few days, thank you for your update, I will try it and let you know.

SamCZ, 2021-11-02:
Hello, just a quck info that it looks like it works now (I cound not find the code you mentioned, so I used just the old I have).
https://cdn.discordapp.com/attachments/703626036503511081/905213562736234506/2021-11-02_22-53-23.mp4

SamCZ, 2021-11-02:
And the problem was not in the voxel mesh having faces on same position, but the trees I generated (there was like 10 polygons on same location per every block)

JulioJerez, 2021-11-03:
ja ja that video loos very cool.
is that rolling tree a compound mesh?

SamCZ, 2021-11-03:
righ now I just used ndShapeBox(0.25f, 0.25f, 0.25f) for the tree. Later I will create compounds from loaded fbx informations

JulioJerez, 2021-11-03:
For a tires like those try using convex hulls, it will behave much nicer
and will be equally fast.

SamCZ, 2021-11-21:
Hi what would you recommend as collider for these type of trees ?
[image]
Compaund of AABB's ? I want players to stand on the branches.

JulioJerez, 2021-11-21:
I world use a compound collision  with each major branch, a convex hull.
should do it.

closed 2022-03-02 by JulioJerez