MADEAPPS/newton-dynamics #251: Cannot compile AVX2 and SSE4 plugin on mingw

ImperatorS79, 2021-08-19:
When trying to compile with SSE4 and AVX2 plugin I get the following errors at the linking step:

```
C:/tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/dgNewtonSse4.2.dir/objects.a(dgNewtonPluginStdafx.cpp.obj):dgNewtonPluginStdafx.cpp:(.rdata$_ZTV12dgConstraint[_ZTV12dgConstraint]+0x28): undefined reference to `dgConstraint::GetMassScaleBody0() const'
C:/tools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/dgNewtonSse4.2.dir/objects.a(dgNewtonPluginStdafx.cpp.obj):dgNewtonPluginStdafx.cpp:(.rdata$_ZTV12dgConstraint[_ZTV12dgConstraint]+0x30): undefined reference to `dgConstraint::GetMassScaleBody1() const'
collect2.exe: error: ld returned 1 exit status
```
Without those plugins everything compiles fine.

JulioJerez, 2021-08-19:
I'd never buidl 3.14 using mingw, can you try 4.0 ?
we has build this Linux, Mingw and window using Clang and intel 
plus at this time 4.00 surpasses 3.14  in quality and almost matches in funtionality.

ImperatorS79, 2021-08-19:
4.0 does not build:
```
C:/tools/mingw-w64-newton-dynamics/newton-4.00/sdk/dCollision/ndShapeStaticProceduralMesh.cpp: In constructor 'dTempArray<T>::dTempArray(dInt32, T*)':
C:/tools/mingw-w64-newton-dynamics/newton-4.00/sdk/dCollision/ndShapeStaticProceduralMesh.cpp:37:3: error: 'm_array' was not declared in this scope
   37 |   m_array = buffer;
      |   ^~~~~~~
C:/tools/mingw-w64-newton-dynamics/newton-4.00/sdk/dCollision/ndShapeStaticProceduralMesh.cpp:38:3: error: 'm_capacity' was not declared in this scope
   38 |   m_capacity = maxSize;
      |   ^~~~~~~~~~
C:/tools/mingw-w64-newton-dynamics/newton-4.00/sdk/dCollision/ndShapeStaticProceduralMesh.cpp: In destructor 'dTempArray<T>::~dTempArray()':
C:/tools/mingw-w64-newton-dynamics/newton-4.00/sdk/dCollision/ndShapeStaticProceduralMesh.cpp:43:3: error: 'm_array' was not declared in this scope
   43 |   m_array = nullptr;
      |   ^~~~~~~
```

JulioJerez, 2021-08-20:
oh yes ther will be an error there.

fixed by doing this
dArray<T>::m_array = buffer;

please sync and try again.
Thank for the bug

ImperatorS79, 2021-08-20:
Still does not work:
```
In file included from C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dCore/dSaveLoadSytem.cpp:26:
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dCore/dFixSizeArray.h: In member function 'void dFixSizeArray<T, maxSize>::SetCount(dInt32)':
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dCore/dFixSizeArray.h:69:12: error: there are no arguments to 'dClamp' that depend on a template parameter, so a declaration of 'dClamp' must be available [-fpermissive]
   69 |  m_count = dClamp (count, 0, maxSize);
      |            ^~~~~~
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dCore/dFixSizeArray.h:69:12: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
make[2]: *** [sdk/CMakeFiles/ndNewton.dir/build.make:986: sdk/CMakeFiles/ndNewton.dir/dCore/dSaveLoadSytem.cpp.obj] Error 1
```

JulioJerez, 2021-08-21:
I see, it seem GCC parses templates functions even when they are not used 
it is thsi funtion 

```
void dFixSizeArray<T, maxSize>::SetCount(dInt32 count)
{
	m_count = dClamp (count, 0, maxSize);
}

```

dClamp is define in dTypes.h but I can no included there because it will generate circular dependency in headers.

I change to this :

```
void dFixSizeArray<T, maxSize>::SetCount(dInt32 count)
{
	m_count = (count < 0) ? 0 : ((count > maxSize) ? maxSize : count);
}

```

which is the same thing.
please try again see if ther are not more errors

ImperatorS79, 2021-08-21:
So I had to modify this file `newton-4.00\sdk\dNewton\ndDynamicsUpdateOpencl.cpp`:
```
@@ -83,7 +83,7 @@ class dOpenclBuffer: public dArray<T>
	{
		void* const source = &(*this)[0];
		cl_int err = clEnqueueReadBuffer(commandQueue, m_gpuBuffer,
-			CL_FALSE, 0, sizeof(T) * GetCount(), source,
+			CL_FALSE, 0, sizeof(T) * dArray<T>::GetCount(), source,
			0, nullptr, nullptr);
		dAssert(err == CL_SUCCESS);
	}
@ @@
```

and now I have:
```
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp: In member function 'void ndDynamicsUpdateOpencl::CopyBodyData()':
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:684:19: error: 'union cl_float4' has no member named 'x'
  684 |   data.m_rotation.x = body->m_rotation.m_x;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:685:19: error: 'union cl_float4' has no member named 'y'
  685 |   data.m_rotation.y = body->m_rotation.m_y;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:686:19: error: 'union cl_float4' has no member named 'z'
  686 |   data.m_rotation.z = body->m_rotation.m_z;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:687:19: error: 'union cl_float4' has no member named 'w'
  687 |   data.m_rotation.w = body->m_rotation.m_w;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:688:19: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'x'
  688 |   data.m_position.x = body->m_matrix.m_posit.m_x;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:689:19: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'y'
  689 |   data.m_position.y = body->m_matrix.m_posit.m_y;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:690:19: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'z'
  690 |   data.m_position.z = body->m_matrix.m_posit.m_z;
      |                   ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:691:16: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'x'
  691 |   data.m_veloc.x = body->m_veloc.m_x;
      |                ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:692:16: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'y'
  692 |   data.m_veloc.y = body->m_veloc.m_y;
      |                ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:693:16: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'z'
  693 |   data.m_veloc.z = body->m_veloc.m_z;
      |                ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:694:16: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'x'
  694 |   data.m_omega.x = body->m_omega.m_x;
      |                ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:695:16: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'y'
  695 |   data.m_omega.y = body->m_omega.m_y;
      |                ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:696:16: error: 'cl_float3' {aka 'union cl_float4'} has no member named 'z'
  696 |   data.m_omega.z = body->m_omega.m_z;
      |                ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:697:18: error: 'union cl_float4' has no member named 'x'
  697 |   data.m_invMass.x = body->m_invMass.m_x;
      |                  ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:698:18: error: 'union cl_float4' has no member named 'y'
  698 |   data.m_invMass.y = body->m_invMass.m_y;
      |                  ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:699:18: error: 'union cl_float4' has no member named 'z'
  699 |   data.m_invMass.z = body->m_invMass.m_z;
      |                  ^
C:/tools/mingw-w64-newton-dynamics/newton-dynamics/newton-4.00/sdk/dNewton/ndDynamicsUpdateOpencl.cpp:700:18: error: 'union cl_float4' has no member named 'w'
  700 |   data.m_invMass.w = body->m_invMass.m_w;
      |                  ^
```

JulioJerez, 2021-08-21:
al right fixed.
Noticed that the OpenCL solver is not completed and I am not planning to supported.
There are many reasons for that, teh tow main ones:
1-two difficult and cumbersome and very hard to maintain.
2-has been abandoned by the major hardware makers.   
3-The standards is in a state of disarray,  where some makes supports some features and other do not.   
4-many more, that are just more aggravations, like lack of default cpu fallback support.  

Instead I am waiting a little for https://www.khronos.org/sycl/
this is like the generalization of an idea Microsoft introduced with VS 2012 called AMP them deprecated. 
this will allow for coding Parallel kernels in that can run in CPU, GPU and even both at the same time.
has the convenience of coding in the same CPP language and has made very good progress in the last few years.
Microsoft has not adopted in visual studio yet  but since they now let people use other compilers, then that's not a problem.
so when it is that time again we will simply replace the OpenCl solver with the  SYCL solver. I just let it there for reference.

Please do not try to select the OpenCL solver, is is the only one will not work, the others do work.

Julio.

ImperatorS79, 2021-08-21:
It still does not compile, same error.

JulioJerez, 2021-08-21:
ah yes my mistake, OpenCL in GCC does no define macro
__CL_HAS_ANON_STRUCT__

so here
```
typedef union
{
    cl_float  CL_ALIGNED(16) s[4];
#if __CL_HAS_ANON_STRUCT__
   __CL_ANON_STRUCT__ struct{ cl_float   x, y, z, w; };
   __CL_ANON_STRUCT__ struct{ cl_float   s0, s1, s2, s3; };
   __CL_ANON_STRUCT__ struct{ cl_float2  lo, hi; };
#endif

``` 
so I have to use s[] instead of x, y, x
please try again. sorry about that, my mistake.

ImperatorS79, 2021-08-21:
If I add the `dArray<T>::` in the OpenCL file now it compiles

JulioJerez, 2021-08-21:
what Lines? there are many place where the array is used.

JulioJerez, 2021-08-21:
oh, I think I found it.
fixed.

closed 2022-03-02 by JulioJerez