MADEAPPS/newton-dynamics #254: Please make a new tag for both 3.14, and 4.0 release series

jonesmz, 2021-09-11:
Ideally, the 4.0 branch can be changed so that it no longer has the 3.14 folder at all...

JulioJerez, 2021-09-12:
eventually I will make a new repository for 4.0, and remove it for the current one 
but I am planning to keep 3.14 as it for a while, the majority of users are still in 3.14 and I get request to bug fixes or minor updates. 
the two directories are 100% independent of each other, and 3.14 is jut a few 10's of meg of extra space that does harm anything.
I do not really see the need to remove it now, or to place on another repository.

jonesmz, 2021-09-12:
The whole point of a version control system like git is that switching between branches is trivial.

Putting 3.14 and 4.0 as folders in the same branch causes, at the very least, a lot of confusion.

JulioJerez, 2021-09-12:
well, we disagree on that point.
I made that change now I will be making the small user base protest because nwo they will have to go some where to get any update. so people has projects for then time is valuable. 

liek I said when it is time, I will move 4.0 to it own branch or own github download. now it is not that time.,

jonesmz, 2021-09-12:
Buy what do you mean they will have to go somewhere?

Just have the 3.x code in a 3.x branch, and the 4.x branch in a 4.x branch.

The only thing end users will have to do to switch is type "git checkout 3.x" or "git checkout 4.x" depending on which version they want.

100espressos, 2021-11-08:
Each version in its own [branch](https://git-scm.com/docs/git-branch) would be typical, allowing both branches to use the folder name 'newton', and allowing users to avoid transferring two repos when they want only one.

As perhaps a bit of workaround, one can add to .gitignore something like `newton-3.14/`, and can setup a 'git directory template' to apply it automatically before cloning.
🌹

jonesmz, 2022-05-07:
4.0 has a tag now, so this issue can be closed.

The whole "All of 3.0 is a folder inside the 4.0 tag" issue is still present, but the request for an official tag in git is satisfied.

closed 2022-05-07 by jonesmz