MADEAPPS/newton-dynamics #255: better Cmakelists structuring

tigercoding56, 2021-09-12:
in the cmakelists txt there is a line if not android add option -msse   this     assumes that the user uses  a compatible architecture which sometimes they do not    so maibe you could replace it with if not architecture == ARM 
    add option -msse

JulioJerez, 2021-09-12:
Does Arm supports -MSS3

I thought is was neo intrinsics.

tigercoding56, 2021-09-13:
i'll just close this for now  because im kind of getting 

> Does Arm supports -MSS3 I thought is was neo intrinsics.
> […](#)

i dont think it supports -MSS3    
and with the issue i  assumed that if it compiles on android it'll compile on other platforms so why not use  architecture = arm instead of is os android if  i am wrong please  close this issue