MADEAPPS/newton-dynamics #258: 4.0 Assert ndArray

kklouzal, 2021-12-08:
[image]
Assert during call to update world.
Call stack:
>	ndNewton_d.dll!dArray<ndBodyKinematic *>::operator[](int i) Line 114	C++
 	ndNewton_d.dll!ndDynamicsUpdate::Update() Line 2031	C++
 	ndNewton_d.dll!ndWorld::SubStepUpdate(float timestep) Line 560	C++
 	ndNewton_d.dll!ndWorld::ThreadFunction() Line 476	C++
 	ndNewton_d.dll!ndWorldDefaultScene::ThreadFunction() Line 54	C++
 	ndNewton_d.dll!dThread::ThreadFunctionCallback() Line 111	C++
 	[External Code]	
This was using the latest code commit https://github.com/MADEAPPS/newton-dynamics/commit/21980c1b9638f2d8694081ae08881b00061de3fa

JulioJerez, 2021-12-08:
the trace stack does not say much. 
did this happen is the sandbox demos.
My guess is that I probably commit a work in progress 

I am trying to add some solver optimization and refactoring, and may that solve has some debug test, but that souel never happens. give me some time to verify that the current commit is right.

kklouzal, 2021-12-08:
Yes same error here happens trying to run test cpp. Probably accidently committed a work in progress :)
I have gone back to using a previous commit for now.

JulioJerez, 2021-12-08:
Ah OK, I will check it out tonight.

JulioJerez, 2021-12-11:
ok I just merged the changes that for the new solver and as I suspected the `assert` was caused by some debug code I added to compare the result of the new solver to the old solve, I have too, by after commenting it out is not longer happening 
this code

```
//xxxx++;
//if (xxxx == 100)
//{
//	const ndArray<ndBodyKinematic*>& x = m_world->GetScene()->GetActiveBodyArray();
//	ndBodyKinematic* body = x[4];
//	body->m_equilibrium = 0;
//	body->m_veloc.m_z = 10.0f;
//}
```

JulioJerez, 2021-12-11:
my guess is that all those other asserts are generated by the solver assuming is running a specific test demo, 
If you sync the bug should go away.

after you sync you will have to make some changes since I renamed some of the base classes to have the nd prefix instead of teh d which can conflict with some classes in 3.14

kklouzal, 2021-12-11:
Yes this has been resolved with the latest sync. Great job thank you!

closed 2021-12-11 by kklouzal