MADEAPPS/newton-dynamics #261: Assert ndShape

kklouzal, 2021-12-11:
[image]
```
>	ndNewton_d.dll!ndShape::~ndShape() Line 54	C++
 	ndNewton_d.dll!ndShapeNull::~ndShapeNull() Line 58	C++
 	ndNewton_d.dll!ndDummyCollision::~ndDummyCollision() Line 47	C++
```
Not sure about this one, started happening with the latest commit https://github.com/MADEAPPS/newton-dynamics/commit/757c1dbd91c0dd7fa9cecc5579584020afece6b6 Is pretty bad, locks my entire system up when it happens with graphics driver crash.

JulioJerez, 2021-12-11:
are you trying to delete a shape from a body?  
when a body is created, he gets a NullShape, the Null Shape is a singleton.  you should not delete it

In any case that was a legacy form 3.14, I now made the shape allocate form eth free lists, so the get a allocated NullShape. 
this does not matter because since the shape will be replaced, it will go to the free list and next shape will get the same memory.

please Sync and try again.

kklouzal, 2021-12-11:
I think this is good now, it hasn't happened again. I will close the issue in a few hours if it stays okay. Thank you.

closed 2021-12-12 by kklouzal