MADEAPPS/newton-dynamics #262: Replace/Integrate GLM as mathematics backend

kklouzal, 2021-12-12:
### I propose replacing the current maths implementation with OpenGL Mathematics (GLM).

**GLM is a very well established, feature rich, highly performant, header only, mathematics library.**

A direct quote from their website:
_"This library works perfectly with OpenGL but it also ensures interoperability with other third party libraries and SDK. It is a good candidate for software rendering (raytracing / rasterization), image processing, physic simulations and any development context that requires a simple and convenient mathematics library."_

Why?
- Reduce Newtons overall codebase making it easier to manage, maintain, and improve upon in the future.

- Newton will more tightly couple together with and integrate into applications. Functions will accept arguments and return values that are directly interchangeable with applications that also use GLM for their mathematics.

- Performance. GLM boasts performance overall providing native internal SIMD support on mathematical operations. If your compiler boasts the AVX flags, GLM will also detect this and compile using AVX optimizations where possible.

- GLM is stable, future versions and updates promise to not break current implementations.

- C98 compatibility means direct integration into Newton without the need to completely restructure the library back-end.

- GLM has a huge community of contributors dedicated to improving and maintaining the library which offers Newton a top-of-the-line mathematics library that is cross-platform. highly stable, feature rich, and bug free.

- GLM variable types are built to work hand-in-hand with GLSL shader code making the push to GPU processing even that much easier with Vulkan.

**While I understand this would require a substantial amount of upfront work, the end result will provide many enhancements to the speed, usability, maintainability, and future improvement of Newton.**

Main cause for action:

1. Applications already using GLM for their mathematics will have a 1:1 relationship calling functions and returning values from/to Newton/parent application. No need to constantly convert back and fourth.

2. No need to waste valuable time creating, maintaining, worrying about a separate mathematics library. Just use GLM and have peace-of-mind knowing you have a highly stable, performant, feature rich mathematics implementation backed by hundreds of developers pushing constant updates and improvements.

3. GLSL compliant variable types. When the time comes to implement Newton on the GPU with Vulkan, types offered with GLM will be easily passed back and fourth between CPU/GPU removing a layer of complexity that you might otherwise be up against.

**I urge you to take time to seriously consider the pros and cons of such a change as I know this will benefit Newton in more ways than we could imagine.**

https://github.com/g-truc/glm

JulioJerez, 2021-12-12:
I appreciate the recomendation,  GLM is very nice and feature rich
but at this time, we are sticking with our version. 
It provides what we need with little overhead.  there is also some functionality in ndMath 
that is not provide by libraries like GLM. 
Also newton is not an openGL application.

Hurleyworks, 2021-12-12:
Here's another interesting library that you might want to check out for the
future

https://github.com/mitsuba-renderer/enoki

kklouzal, 2021-12-12:
Kind of thought so :)

Just wanted to throw the idea out there for you to consider, personally I see a lot of benefit can be gained from making a move like this, but then again nobody knows the intricacies of the library like you do.

For you to not have to worry about maintaining the maths backend I figure would take a large chunk of the development overhead off your plate. The interoperability aspect we gain from making the change would be very nice too!

So just something to keep in the back of your mind!