MADEAPPS/newton-dynamics #263: Unresolved Externals Inheriting From ndBodyDynamic

kklouzal, 2021-12-13:
Inheriting from `ndBodyDynamic`:
`class TriangleMeshSceneNode : public SceneNode, public ndBodyDynamic`

Results in unresolved externals:
```
1>main.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl ndBodyDynamic::IntegrateGyroSubstep(class ndVector const &)" (?IntegrateGyroSubstep@ndBodyDynamic@@UEAAXAEBVndVector@@@Z)
1>main.obj : error LNK2001: unresolved external symbol "public: virtual class ndJacobian __cdecl ndBodyDynamic::IntegrateForceAndToque(class ndVector const &,class ndVector const &,class ndVector const &)const " (?IntegrateForceAndToque@ndBodyDynamic@@UEBA?AVndJacobian@@AEBVndVector@@00@Z)
```

I can get around it by defining those functions inside my class that is inheriting from `ndBodyDynamic`:
```
class TriangleMeshSceneNode : public SceneNode, public ndBodyDynamic
{
public:
...
	inline void IntegrateGyroSubstep(const ndVector&)
	{
	}

	inline ndJacobian IntegrateForceAndToque(const ndVector&, const ndVector&, const ndVector&) const
	{
		ndJacobian step;
		step.m_linear = ndVector::m_zero;
		step.m_angular = ndVector::m_zero;
		return step;
	}
...
}
```
I just copied them out of `ndBodyKinematic.h` but I think these virtual function overrides are _supposed_ to be optional..

kklouzal, 2021-12-13:
Well, my dynamic bodies fall through the world now. They interact with my kinematic objects but no longer interact with other dynamic objects..

kklouzal, 2021-12-13:
I used the functions inside ndBodyDynamic.cpp and interactions returned.
Not sure why Visual Studio thought ndBodyKinematic.h was the appropriate place to grab the functions from, might be related to why it complains about unresolved externals. Nonetheless I don't think it should be required to override these functions to inherit from ndBodyDynamic..

JulioJerez, 2021-12-13:
yes, you are correct those functions should be private member of ndBodyDynamic
fixed, please try again.

noticed that ndBodyKinematic is the base class of ndBodyDynamic
these are part of the world but they are not static bodies. 
In newton, it is the rule that a static body is a dynamics body with infinite mass, which in newton is define as mass set to zero.

ndBodyKinematic can be part of the world, but they are object that do no interact with other kynematic bodies.  they are used for game play stuff like player capsule, trigger volumes, late the fluids will be wrapped on a kinematic body. 
the purpose of them to be part of the world is that that the solve is aware of them and they can act on rigid bodies, but it is one way interaction.

kklouzal, 2021-12-19:
Verified this is working as intended on latest commit.
Thank you! :)

closed 2021-12-19 by kklouzal