MADEAPPS/newton-dynamics #274: Newton update for "Amnesia: The Dark Descent"

Spatzendame, 2022-02-09:
Hi,

I want to update the newton.dll for this game. But when I replace the old original file with the latest version and start the Launcher.exe an error message occurs. Can someone here tell me why that happens and whether it is even possible to upgrade this library and how to fix it/make it compatible?

Thank you so much.

JulioJerez, 2022-02-10:
I do not think they are binary compatible.

Spatzendame, 2022-02-10:
Hmm, okay. But there's any way to make it compatible? I mean we have both source codes from Amnesia and Newton.

Hedede, 2022-03-28:
> Hmm, okay. But there's any way to make it compatible? I mean we have both source codes from Amnesia and Newton.

Compile them both together?

JulioJerez, 2022-03-28:
well, It should be very easy thing to do for a programmer with a better that beginner skill set, 
the new library of far, far better. faster and more accurate. But I do not have time.
beside I think the Amnesia license is GPL, so will be problematic for me to do it.

If you or someone try to do it, I can help but giving some pointers, but I cannot be a lead. 
I would love to see that game with the new engine.

Spatzendame, 2022-03-29:
Amnesia is open source here on Github. 

https://github.com/FrictionalGames

So a programmer should be able to compile new dlls for it very easily. But my skills are way too low.

antonyvolkoff, 2023-01-22:
i updated the library to version 2.36 but i get assert in collision detection when calculating support vertices.
if the library is in release, I don’t get the assert, but the character falls down
I can't solve this problem. I don't understand the calculations inside the library
`_ASSERTE(dgAbsf(dir % dir - dgFloat32 (1.0f)) < dgFloat32 (1.0e-3f));`

JulioJerez, 2023-01-22:
It mean some matrix is not orthonormal.

You updated to a very old version, why didn't you go for 3.14?
The interface is almost identical.
2.xx is not even if GitHub. So it is hard to make any fix.

JulioJerez, 2023-01-22:
You can put a break point and see if the magnitude of fire is not too far
from one, and just comment out the assert. Si if that works.