MADEAPPS/newton-dynamics #278: FATAL ERROR: 'result_of': is not a member of 'std'

kklouzal, 2022-05-13:
My application requires c++20 and std::result_of has been depreciated as of c++17 and removed in c++20 and beyond.

https://github.com/microsoft/STL/pull/380

Error	C2039	'result_of': is not a member of 'std'	 - C:\Libraries\newton\include\ndNewton\ndThreadPool.h	163	

As a result, Newton will not compile..

JulioJerez, 2022-05-13:
ah one more changes, in the pass when I was no using lambdas function for thread kernels, I have to have a variable argument 
operator ()
because that was the only way to pass parameter to the jobs, it was complicated because each job was a class that has to be created and them has to have copy constructor, and got more a more complex. 

the when IO changed to lambdas, I was still parameters parameter as argument so that did no changed, is anything, it made harder to initialized the jobs. 
them I learned that lambdas function are not unnamed functions, in fact there are unnamed function when the function does no declare closure argument. when using closures, a lambda is a object that is created on the stack  by the compiler.
so the same technique I was using before of initialize the class with copy constructor because much elegant, 
so I made that template function to do the initialization automatically as I was doing originally.
```
namespace ndMakeObject
{
	template<typename Type> auto ndFunction(const Type & obj) -> decltype (::ndFunction<Type>(obj))
	{
		return ::ndFunction<Type>(obj);
	}
}

```

this removed all the variable arguments, and the only parameter that the kernel need is the thread index and the thread count.
everything else  is pass by the closure of the template and that copy constructor above. the onle part I forgut was to re write the operator () so that is does no need those template arguments anymore. 
the code now reduces to this

```
	//template<typename... Args> typename
	//std::result_of<const Type(Args...)>::type operator()(Args... args) const
	//{
	//	m_object.operator()(args...);
	//}

	void operator()(ndInt32 threadIndex, ndInt32 threadCount) const
	{
		m_object.operator()(threadIndex, threadCount);
	}

```

this is a tiny optimization and far more clean, elegant and cross platform.
if you sync, you will get the stream lined version.

thanks for the report.

Julio

kklouzal, 2022-05-13:
Thank you for the quick update :) keep up the good work

closed 2022-05-13 by kklouzal