MADEAPPS/newton-dynamics #279: FATAL: Division By Zero

kklouzal, 2022-05-13:
As of this commit: https://github.com/MADEAPPS/newton-dynamics/tree/31e4d0d43c956e22e79c1e67567fe9c0bccdbaf0 I am getting a division by zero crash:

Unhandled exception at 0x00007FFB3219B0C0 (ndNewton.dll) in Vulkan.exe: 0xC000008E: Floating-point division by zero (parameters: 0x0000000000000000, 0x0000000000009D04).

Call Stack:
>	[Inline Frame] ndNewton.dll!ndDynamicsUpdate::InitWeights() Line 614	C++
 	ndNewton.dll!ndDynamicsUpdate::Update() Line 1501	C++
 	[Inline Frame] ndNewton.dll!ndWorld::SubStepUpdate(float) Line 491	C++
 	ndNewton.dll!ndWorld::ThreadFunction() Line 426	C++
 	ndNewton.dll!ndThread::ThreadFunctionCallback() Line 111	C++

JulioJerez, 2022-05-13:
that's should neve happens.  are you setting the CUDA solver, that's no ready yet. it only work on that test case. 
I will be some time until it can handle joints. 
I just try few scene and I did not see that error happening. 
does that happen in the sandbox demos?

kklouzal, 2022-05-13:
Not using CUDA or AVX solver.
```
//
//	Physics Initialization
_ndWorld = new ndWorld();
_ndWorld->SetThreadCount(std::thread::hardware_concurrency() - 2);
_ndWorld->SetSubSteps(3);
_ndWorld->SetSolverIterations(2);
```
And no it doesn't appear to be happening in the sandbox demo. All I've done here is update from an older version of Newton, not sure where to look for what's wrong..

kklouzal, 2022-05-13:
No, i'm sorry, i've fixed it.
`_ndWorld->Update(deltaFrame);`
When the application first starts, `deltaFrame` is `0` and that was causing the issue. I have simply added a check for this and we're good to go!
```
if (deltaFrame > 0.0f)
{
	_ndWorld->Update(deltaFrame);
	SceneGraph::updateUniformBuffer(currentFrame);
	//
	//	Draw Frame
	Render();
}
```

closed 2022-05-13 by kklouzal

JulioJerez, 2022-05-13:
oh, I was going to say the only divide on the crash is 1 / timestep.  
so maybe some how the app is passing a time step of zero. 

but you found it before I made the comments. so it should be cool now.
 
One note, I see you are using function std::thread::hardware_concurrency()
that is cool, but one of the thing the engine is that when using hyperthreaded, does no really make much differences since when dealing with large scenes, it is all dominated by memory bandwidth 

I have seen occasions where setting many thread has a negative impact once it start using more thread than actual cores. 
it s my experiences, it is better to use about half the number of thread so that the OS does the core assignment  but still have enough thread for the rest of the systems.

basically what happen is that many threads cause that each tread has a lighter work load,  so when a thread complete it task it has to do a preemption and those are no cheap.

I have struggle for years to reduces that, and does no seem to be a solution no a lest of pc os. 

in fact I found that windows is in fact the most lenient OS, of all, all other Linux and OSX task switiching is a ferociously expensive.  so what I found is that a Good rule of thumb is to use a number of thread equal to the number of actual cores, so that each thread get large enough task that consume a good part of the time slice. 

more thread cause the OS to do more task switching, and that time goes to waste.