MADEAPPS/newton-dynamics #280: Problems with 3.14

KionX, 2022-05-28:
NewtonBodySetCollidable(Body,1); disables collisions for KinematicBody.

Also each NewtonWorld creates threads. It should use a thread pool.
I have many worlds. And Newton creates a total of about 100 threads.

JulioJerez, 2022-05-29:
> NewtonBodySetCollidable(Body,1); disables collisions for KinematicBody.

the function only acct of the body that is called for. 1 should make it collidable,  0 no collidable.  0 is the default.
that is the expected behavior.

> Also each NewtonWorld creates threads. It should use a thread pool.
> I have many worlds. And Newton creates a total of about 100 threads.
> 
tow instance of the engine are independent of each other, so thread are part of the engine. 
the engine run on it own thread which is part of it.

if you are using so many worlds, then maybe for you should disable threading. 
there is a define in the cmake script "NEWTON_BUILD_SINGLE_THREADED" that builds the engine single thread.  
check that, and them the engine will run on the application thread. 
them you can make your own thread pool and send the update as task, or you can run it however you like.

finally:
Newton 4. has been released for a while and is very stable, more accurate and a lot easier to use that 3.14
so if you are no too far alone, try integration 4 you can make so that them can coexist until you  migrate to 4

KionX, 2022-05-29:
**1 should make it collidable**
Now it works the other way around.

My worlds sleep most of the time.
I don't have the option to upgrade to Newton 4 right now.