MADEAPPS/newton-dynamics #282: Manually setting / adding rotation to Dynamic Body

kered59, 2022-06-20:
Hello all, 

Currently I have been tinkering around with Newton Dynamics for a few months but have not found a method to easily rotate Dynamic Body without directly manipulating the Matrix itself.

Is there a method to directly manipulate the rotation of a Dynamic body and if not has anyone found a work around to this or perhaps a easy solution that I could utilize moving forward?

Thanks in advance.

JulioJerez, 2022-06-21:
What are you trying to do?

kered59, 2022-06-21:
I am trying to directly set the rotation of an object without applying forces.
If needed I can provide an example.

kered59, 2022-06-21:
Thank you for your input, could you give a brief example here of how to use the joints for this?
In case someone else comes across the same issue.

Setting the rotation directly is a reference to providing a vector of rotational values and having the object adopt that rotation directly without the need for extra steps. I had a method to manipulate the matrix directly but am lacking in capability to do it efficiently at this time hence asking if there were any alternatives.

Greatly appreciate the input!

JulioJerez, 2022-06-21:
in the demos file
..\newton-dynamics\newton-4.00\applications\ndSandbox\ndDemoCameraManager.cpp
line 275

is the code that makes a joint and attaches it to a body.

     m_pickJoint = new ndDemoCameraPickBodyJoint(body,
> scene->GetWorld()->GetSentinelBody(), posit, this);
> scene->GetWorld()->AddJoint(m_pickJoint);
> m_pickingMode ?
> m_pickJoint->SetControlMode(ndJointKinematicController::m_linear) :
>
> m_pickJoint->SetControlMode(ndJointKinematicController::m_linearPlusAngularFriction);

m_pickJoint->SetMaxLinearFriction(mass.m_w * linearFrictionAccel);
> m_pickJoint->SetMaxAngularFriction(inertia * angularFritionAccel);

the the Joint has functions for setting the transform

kered59, 2022-06-22:
Big thanks again!

closed 2022-06-22 by kered59