MADEAPPS/newton-dynamics #287: NEWTON_DOUBLE_PRECISION=on broken since f8e1a3c64b70dff7159416e3a8abb7874f547783

perkinslr, 2022-07-10:
Starting with that commit, building with double precision results in
```
/tmp/newton-dynamics/newton-4.00/sdk/dCollision/ndScene.cpp:2816:13: error: no matching function for call to 'ndMax'
                                max_x = ndMax(intPosit.m_ix, max_x);
```
(and the same on the next 2 lines)

And as of commit a5f11c2feec9383e4c04dde2a855c309371a2eee
it also has the following error
```
/tmp/newton-dynamics/newton-4.00/sdk/dCollision/ndScene.cpp:96:5: warning: array index 3 is past the end of the array (which contains 3 elements) [-Warray-bounds]
                                m_keyCode[3] = 2;
                                ^         ~
/tmp/newton-dynamics/newton-4.00/sdk/dCollision/ndScene.cpp:105:4: note: array 'm_keyCode' declared here
                        ndUnsigned32 m_keyCode[3];
```

JulioJerez, 2022-07-10:
Ah I will fix it, but I can't commit at the moment since I am moving that
functionality to it own file, since it has become more complex.

The is part of the change to the scene manager for gpu.
Basically the build has to be what they call embarrass parallel,
essentially lock and atomic free, and linear scaling.

At first I thought that the gpu will not need it but in practice there are
object that are just to hard to do in gpu, stuff like find joint, raycast,
convex case, and so on, you can offload to gpu because the gou sence is few
frame behind.

So this imply that the cpu need to keep a spatial scene fir the client app
doing queries.

So I start from there. But is has been quite challenging to make and
incremental bvh tree, that is embarrassing parallel.
Hourly I think I have now.

Any way, I will check the double build.

JulioJerez, 2022-07-11:
ok the double precision problem is fixed now.

closed 2022-10-08 by JulioJerez