MADEAPPS/newton-dynamics #307: 4.02 EM_INVALID not defined when building MinGW-w64

brechtsanders, 2022-12-31:
I was trying to build version 4.02 on Windows using MinGW-w64 and got errors that `EM_INVALID` is not defined.

It looks like it is referenced in `sdk/dCore/ndUtils.h`:
```c
	#if defined (WIN32) || defined(_WIN32)
		#define D_FLOAT_EXCEPTIONS_MASK	(EM_INVALID | EM_DENORMAL)
	#else
		#define D_FLOAT_EXCEPTIONS_MASK	(FE_INVALID | FE_INEXACT)
	#endif
```
The problem is the `#if` line assumes Windows builds are always done with MSVC.

Changing that line like this allows the build to continue:
```c
	#if (defined (WIN32) || defined(_WIN32)) && !defined(__MINGW32__)
```

But you may want to change it to checking something more MSVC specific.

JulioJerez, 2022-12-31:
fixed, thanks 
I made liek this 

	//#if defined (WIN32) || defined(_WIN32)
	#if defined (_MSC_VER)
		#define D_FLOAT_EXCEPTIONS_MASK	(_EM_ZERODIVIDE | _EM_INVALID | _EM_DENORMAL)
	#else
		#define D_FLOAT_EXCEPTIONS_MASK	(FE_DIVBYZERO | FE_INVALID | FE_INEXACT)
	#endif

in any case, the exceptions are only valid in windows system, all othe OS ignored it.
This is good for debugging and to get a clean code with hard to find float bugs, 
but Later I will add an option to the build system that disable them. 
production code should not required  float exceptions.

Thanks for the report

brechtsanders, 2023-01-02:
That's exactly why I stay far away from MSVC: to avoid Microsoft's sometimes deviating standards.
Your solution works for me, I just tried it and it builds with MinGW-w64.
Thanks!

closed 2023-01-02 by brechtsanders