MADEAPPS/newton-dynamics #308: Odd trigger behavior

lperkin1, 2023-02-18:
I don't know if I'm missing something, but I was getting some odd callbacks from triggers using newton 4.  So I modified the trigger demo to have the box spawn well above the pool, removed the other objects, and added a log of entry/in/exit callbacks.  When I run that, here's what I get.

```
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
in trigger body: 3
in trigger body: 3
in trigger body: 3
in trigger body: 3
in trigger body: 3

```
It starts with invoking the exit callback and It doesn't call the enter callback.  In my project, it looks like the trigger starts invoking on the aabb overlap, but I haven't dug into that further in the demos.

If in the demo I grab the box and pull it up out of the pool, I get this

```in trigger body: 3
in trigger body: 3
in trigger body: 3
in trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
exit trigger body: 3
in trigger body: 3
```

Ending the demo with the box outside the pool, the last callback is OnTrigger.

Have I found a bug or am I missing something?

JulioJerez, 2023-02-18:
Thanks for the report.

You may have found a bug, later today I will try to recreate the bug.
And try to fix it.

JulioJerez, 2023-02-19:
Please Sync and try again.

lperkin1, 2023-02-19:
That fixed the trigger problem.

Looks like a couple macro changes were causing problems with the swig wrapper in C# so I sent a pull request for that.  There also seems to be a problem with compiling the demos using double precision, but I don't need those unless I find some other bug to test.

Thanks!

closed 2023-02-19 by lperkin1