MADEAPPS/newton-dynamics #310: Comparison with Bullet/Jolt/PhysX

SirLynix, 2023-03-15:
Hi!

I'm interested in comparing Newton Dynamics with [Bullet](https://github.com/bulletphysics/bullet3), [PhysX](https://developer.nvidia.com/blog/announcing-nvidia-physx-sdk-5-0/) and more recently [Jolt](https://github.com/jrouwe/JoltPhysics). Do you have benchmarks (or know some benchmarks), or tests?

I'm interested in performance, stability, usage of multi-core processors (and GPU) and multiplatform support.

Can you tell me how Newton performs (or is expected to perform) compared to those physics engine, or even better if there's a recent demo comparing them?

Thank you.

JulioJerez, 2023-03-15:
I do not spend type on comparison. 
Newton is hobbies project probably slower, than all those you listed.

DaveGravel, 2023-03-15:
Newton may be slower in some cases, but it provides much greater precision in several other cases. However, determining the actual difference cannot be done through a benchmark alone. One needs to use the software to observe the true variance. Furthermore, I must emphasize that the Newton library's code interface is much more well-organized and better written than that of other libraries. This makes it considerably easier to use for professional and highly accurate simulations. If your aim is to simulate thousands of objects and physical precision is not critical, then Newton can certainly meet your needs, although it may not be the optimal choice. On the other hand, if you require a highly precise simulation and the number of objects is not a significant factor, Newton is one of the best options available, in my opinion, if it matters.

SirLynix, 2023-03-17:
As I wrote I'm not only interested in performances but in simulation stability too.

And honestly, after experimenting with Bullet for a while, I think Newton easily outclasses it, I was curious if benchmarks existed but I suppose I'll have to make my own.

Keep the hard work!

TrevorCash, 2023-03-18:
I have done comparisons against bullet myself.  Newton is a much more stable simulation than bullet for sure.  Newton also has very good joint robustness for a given iteration count.  Very good runga kutta solver.  And CPU threading scales very well.  

Using Newton in a project will throw asserts during if you are using it wrong while other libraries might still run.