MADEAPPS/newton-dynamics #311: ndJointList oddity

lperkin1, 2023-05-03:
I don't know if this is supposed to be this way, but the ndJointList classes seem odd to me.

ndBodyKinematic has an internal ndJointList class inheriting from ndList<ndJointBilateralConstraint*>.  
ndWorld uses the class from the ndJointList.h file which inherits from ndList<ndSharedPtr<ndJointBilateralConstraint>>
ndJointBilateralConstraint has an internal class inheriting from ndList<ndSharedPtr<ndJointBilateralConstraint>>

My situation is I'm trying to remove all the joints from a body, without removing the body, in between world update calls.  I had been looping over the list from 
  ndBodyKinematic::GetJointList() 
and calling 
  ndWorld::RemoveJoint(ndJointBilateralConstraint* const joint)

Previously, that call looked up the ndSharedPtr for the joint and removed that from the world.  After I synced yesterday that RemoveJoint overload doesn't seem to be removing the joint from the world's list.  Looks like it just adds the joint to a list of deleted joints.

What am I missing?  Is there something I'm supposed to do to get that list of deleted joints to actually delete, or some way to get the list from the body as ndSharedPtrs, or look up the joint's ndSharedPtr from the world/scene?

JulioJerez, 2023-05-03:
yes, it can be confusing. I need to fix some part of it, but it is not wrong.

if you look at classes Kinematic and file ../newton-4.00\sdk\dCollision\ndJointList.h
they both define a joint list. the class name is the same but they are type of list.

the one in the joint is is a shared pointer use by the world and class ndJointBilateralConstraint to store a node.

the one in the object list, is no shared pointed, because if it was they it will produce circular reference  counting between bodies and joints.

these list are define in separate files because, if they were define on the call that the manage, they in c++ it will generate also circular dependency when the list including the body,  them teh body including the list.

my mistake is that I added the shared pointer a while back but has limited usages, but now after adding the file format, I need to be able to create a scene on a file, that is not added to the world, so that an app can load a scene and the postprocess it, moving the bodies to places, setting the notification, that sort of thing, before adding to the words. 
this will be hard to do not using shared pointer, so I when a head and made usage of the share pointers.

teh secudn mistake is that I left the old list on the ndJointBilateralConstraint  class, but is no used so I just comment out that code.

notice that class ndJointBilateralConstraint  use both type of nodes, so they are properly qualified.

	ndJointList::ndNode* m_worldNode;
	ndBodyKinematic::ndJointList::ndNode* m_body0Node;
	ndBodyKinematic::ndJointList::ndNode* m_body1Node;

JulioJerez, 2023-05-03:
that part one. 

teh second part, removing joints, bodies, and model form the world, will just place them into the delete list, and they will be garbage collected whe the frame update finish. 
this simplify teh scene management a lot, now object can be deleted at any point,  before tshi was problematic, because people delete object form wrong places,  but even if they did not, not having the reference, kept causing destruction order problems. 

the garbage collection happen by teh end of function void ndWorld::ThreadFunction() 

and no that lest is private because those object are supposed to be dead.

JulioJerez, 2023-05-03:
to do this

```
My situation is I'm trying to remove all the joints from a body, without removing the body, in between world update calls. I had been looping over the list from
ndBodyKinematic::GetJointList() 
and calling
ndWorld::RemoveJoint(ndJointBilateralConstraint* const joint)
```

you just delete the joint form the world by calling ndWorld::RemoveJoint(ndJointBilateralConstraint* const joint)

if you has a reference the joint will still be alive, if you don't then teh joint will be deleted at the end of teh frame. 
you can set a break point on ndWorld::ThreadFunction() line 497 you will see that teh joint is deleted.

I hope that helps

JulioJerez, 2023-05-03:
oh I see you confusion. let us go with the basic first. 

I paste the code in the demos as a unit test. 
I made not changes yes, it is just the way you made it.

please sync when you get time, them build it using static libraries (the unit test use static build only)

It should behave the same way, them I will tell you what happens, but as a head up, it is no wrong.

lperkin1, 2023-05-03:
Ok, I've built the test on both the current repo and the previous version and the test runs on both.

Earlier version results:
> [----------] 1 test from RemoveJoints
> [ RUN      ] RemoveJoints.CreateWorld
> startedJoint to remove: 1454095964704
> Next Joint from removed: 1454095964768
> First Joint after remove: 1454095964768

Current version results:
> [----------] 1 test from RemoveJoints
> [ RUN      ] RemoveJoints.CreateWorld
> startedJoint to remove: 1926719815904
> Next Joint from removed: 1926719816640
> First Joint after remove: 1926719815904

Looks like the same behavior as my example running in my project.

JulioJerez, 2023-05-03:
yes that was the expected behavior.
I now added the change to make it work. if you sync it seoul generate this
report

[----------] 1 test from RemoveJoints
> [ RUN      ] RemoveJoints.CreateWorld
> startedJoint to remove: 3034512101312
> Next Joint from removed: 3034512099456
> First Joint after remove: 3034512101312
> Joints in body before tick: 2

> First Joint after remove after next tick: 3034512099456
> Joints in body after tick: 1

this is how it is supposed to work.

the engine main loop is rogfuly liek this.

-Update()
> {
>     -DeleteOld Objects bash on delete lists.
>    -Make a SnapShot of the scene, Get all objects into vector arrays.
>    - update all sybsystes
>   - User feedback
> }

almost all those faces work in parallel on different threads,
therefore changing the scene during update is a complete task that get
increasingly harder as there are more threads and cores.
this is the reason why any object added or deleted during an update, does
see action until the next update.
if objects are added, they go straight to their container lists. and the
container is flagged as dirty.
if objects are removed, they get placed into a delete container.

what is new, is that objects added or deleted from out size the main update
loop are treated exactl;y the same way, and this simply the engine
scene management logic.

and that all there is.

in the test, when the Joint is deleted and a new one is added to the body,
the body will have two joints, but that is fine, because the active joints
are on the working array.
so in that case the number of joints will be 2.

then when an update tick is executed, the garbage collection pass is
performed and the dead body will be removed for the body and safely
deleted, then the number of joints will be 1.

at not time during a simulation update a body will have a duplicated joint.

After you sync, pay attention to the commnet I added, you made a mistake
when getting the list, unfortunately CPP does not enforce the validation of
return values, so it will create a new list.
currently in newton, this generates a crash bug, so to fix it I have to add
a move constructor semantic, but I am not going to do that, simply because
it encourages bad programming.
I am trying to make the engine as functional as possible but I am not
making it idiot proof like Java or Csharp.

anyway thinc and try again, see if is it more clear now.
Julio

lperkin1, 2023-05-03:
Sorry about the bad c++, it's not a language I'm good with.  I'm working on a C# wrapper for Newton 4 to use with Unity so SWIG usually takes care of the c++ for me.

I think I understand the situation at this point.  Obviously this isn't a bug, but if you would indulge me one more question.  In Unity, because I get just one thread for accessing game objects, the server does this during the physics process

> {
>   newtonWorld.Update(Time.FixedDeltaTime);
>   -- other processes including removing joints --
>   -- other processes on remaining joints --
> }

Where the list of joints won't update until the next time Update is called, do I need to just keep track of what joints are dead myself or is there some convenient way to know what joints are still alive before Update cleans up the old objects?

I'm sure I can come up with a (likely bad) way to make it work, but your code is much more efficient than mine.  I'd hate to miss out if there's a good way to get just the live joints at each step in between update calls.

Thanks for your time.

JulioJerez, 2023-05-03:
Ah good question.
You are right. If to iterate over the joint from the world list or from a
body, it is possible that you hit a dead one.

However that should not be a problem, you can operate on the joint because
as long as it is on the list, the joint is active.

But for more completeness, I can make it so that the joint set the is
active flag off when is removed

That's a flag that is part of the joint base class, it is for contact
because there can be many more than on dead c,a..
A  on a body

I will add the functionality to night.

Then, you can just call.

if (Joint.IsActive())
{
     Do something()
}

JulioJerez, 2023-05-04:
ok now the joint will set the active flag when they are bashed for deletion.

you just call 

if (joint->IsActive())
{

}

thsi was already there, I just forgot to set the flag, but teh code do test, for that flag when getting the Joint and contact list into the flat array for the update.

lperkin1, 2023-05-06:
Finally got a chance to run my tests.  Using IsActive works brilliantly.

Thanks for the help!

closed 2023-05-06 by lperkin1