MADEAPPS/newton-dynamics #312: Roller joint limits

lperkin1, 2023-05-20:
The demo for the roller joint doesn't seem to be limiting the object's movement or rotation.

I removed all the other joint examples and the spring damper.  So the object should be limited to (-1, 1).  Here's what I get.

Starting position
[image]

Outside limits
[image]

It's harder to see, but the rotation limits also don't lock the rotation.  Am I missing something?

lperkin1, 2023-05-20:
Looks like I'm getting the same situation with the ndJointCylinder.

JulioJerez, 2023-05-21:
ah my mistake, teh roller joint is very similar to the cylinder, It seemed I copy and paste the code, I do not really remember now, it was so long ago.
The cylinder seem to be correct I just tested.
and I fixed teh bug with the roller.

Please try again.

lperkin1, 2023-05-21:
It's working.  Thanks!

closed 2023-05-21 by lperkin1