MADEAPPS/newton-dynamics #314: Triggers causing assertion fails

lperkin1, 2023-05-23:
Got another one.  Same test using a trigger instead of a kinematic body.  I get an assertion failure [at ndBodyDynamic.cpp(388): count == checkConnection].  I'm guessing that's another check that's obsolete, but here's the test I've got showing the issue.

```
#include "ndNewton.h"
#include <gtest/gtest.h>

class csBodyTrigger : public ndBodyTriggerVolume {

public:
	csBodyTrigger() : ndBodyTriggerVolume() { }
	virtual ~csBodyTrigger() {}
	virtual void OnTrigger(ndBodyKinematic* const body, double timestep) { std::cout << "Trigger\n"; }
	virtual void OnTriggerEnter(ndBodyKinematic* const body, double timestep) { std::cout << "Enter\n"; }
	virtual void OnTriggerExit(ndBodyKinematic* const body, double timestep) { std::cout << "Exit\n"; }

};

TEST(Collisions, TriggerCollision)
{
	ndWorld world;
	ndShapeInstance shapeinst(new ndShapeSphere(ndFloat32(0.5f)));

	ndMatrix matrix(ndGetIdentityMatrix());

	csBodyTrigger* triggerbody = new csBodyTrigger();
	triggerbody->SetCollisionShape(shapeinst);
	triggerbody->SetMatrix(matrix);
	ndSharedPtr<ndBody> triggerPtr(triggerbody);
	world.AddBody(triggerPtr);

	matrix.m_posit.m_y = 5;

	ndBodyDynamic* movingbody = new ndBodyDynamic();
	movingbody->SetNotifyCallback(new ndBodyNotify(ndBigVector(ndFloat32(0), ndFloat32(-9.81f), ndFloat32(0), ndFloat32(0))));
	movingbody->SetCollisionShape(shapeinst);
	movingbody->SetMatrix(matrix);
	movingbody->SetMassMatrix(ndFloat32(10), shapeinst);
	ndSharedPtr<ndBody> movingPtr(movingbody);
	world.AddBody(movingPtr);

	for (int i = 0; i < 480; i++)
	{
		world.Update(1.0f / 60.0f);
		world.Sync();
	}

	world.CleanUp();
}
```

Sorry I can't just figure out what that assertion checks.  With my paltry c++ abilities I can't tell if it's fine to just ignore it or if I've messed something up.  

Thanks for your time.

JulioJerez, 2023-05-23:
The assert are check that I put in the code to verify that some input are
valid.
Also some time is for check a possible case that was not tested.

I will paste that test tonight and see what happens there

That's actually good, because it just improve the reliability of the engine

JulioJerez, 2023-05-23:
In that case may be a side effect of the change I made.
In the pass kinematic bodies where not suppose to be in the scene, but now
they can.

So I need to check the validity when a kinematic body is part of a contact
joint.
By design, they are not part of joints,

So I have to make a better check, something like check for dynamic, them if
fail check if is a kinematic with non zero mass.

I will check that tonight and made a better fix.

lperkin1, 2023-05-24:
Thanks for pointing out the double problem.  I'll go back through my code and fix those.  It took me a while to figure out how your data types work and what I copied was from a while back.

Regardless, I'll keep hunting down my undefined behavior.  Thanks!

lperkin1, 2023-05-24:
Ok, I shouldn't have closed this one quite.  This is working when the body starts outside the trigger.  If a simulation starts with a body already inside the trigger, I get a different assert.

Same test, just commented out the one line setting the dynamic body starting position.

```
/* Copyright (c) <2003-2019> <Newton Game Dynamics>
 *
 * This software is provided 'as-is', without any express or implied
 * warranty. In no event will the authors be held liable for any damages
 * arising from the use of this software.
 *
 * Permission is granted to anyone to use this software for any purpose,
 * including commercial applications, and to alter it and redistribute it
 * freely
 */

#include "ndNewton.h"
#include <gtest/gtest.h>

class csBodyTrigger : public ndBodyTriggerVolume 
{
	public:
	csBodyTrigger()
		:ndBodyTriggerVolume() 
	{ 
	}

	virtual ~csBodyTrigger() 
	{
	}

	virtual void OnTrigger(ndBodyKinematic* const, ndFloat32) 
	{ 
		std::cout << "Trigger\n"; 
	}
	virtual void OnTriggerEnter(ndBodyKinematic* const, ndFloat32)
	{ 
		std::cout << "Enter\n"; 
	}
	virtual void OnTriggerExit(ndBodyKinematic* const, ndFloat32)
	{ 
		std::cout << "Exit\n"; 
	}
};

TEST(Collisions, TriggerCollision)
{
	ndWorld world;
	ndShapeInstance shapeinst(new ndShapeSphere(ndFloat32(0.5f)));

	ndMatrix matrix(ndGetIdentityMatrix());

	csBodyTrigger* triggerbody = new csBodyTrigger();
	triggerbody->SetCollisionShape(shapeinst);
	triggerbody->SetMatrix(matrix);
	ndSharedPtr<ndBody> triggerPtr(triggerbody);
	world.AddBody(triggerPtr);

	//matrix.m_posit.m_y = 5; <--- commenting out this line makes the body start in the trigger

	ndBodyDynamic* movingbody = new ndBodyDynamic();
	movingbody->SetNotifyCallback(new ndBodyNotify(ndBigVector(ndFloat32(0), ndFloat32(-9.81f), ndFloat32(0), ndFloat32(0))));
	movingbody->SetCollisionShape(shapeinst);
	movingbody->SetMatrix(matrix);
	movingbody->SetMassMatrix(ndFloat32(10), shapeinst);
	ndSharedPtr<ndBody> movingPtr(movingbody);
	world.AddBody(movingPtr);

	for (int i = 0; i < 480; i++)
	{
		world.Update(1.0f / 60.0f);
		world.Sync();
	}

	world.CleanUp();
}
```

I get
`newton-4.00\sdk\dCollision\ndScene.cpp(412) : Assertion failed: contact->m_isIntersetionTestOnly`

I guess I'm testing all the combinations with my project :)

JulioJerez, 2023-05-25:
the assert was there just to check. It should just call on enter.

Fixed, thanks

lperkin1, 2023-05-25:
Ok, finally got my trigger tests working.  Fixing all the outdated asserts left me with an assert failure calling out a problem in my code.  Now that's fixed this all looks like it works properly.

Thanks!

closed 2023-05-25 by lperkin1