MADEAPPS/newton-dynamics #315: Large bodies assertion fails

lperkin1, 2023-05-25:
Ok, got another one.  This one is probably me doing something stupid though.  Now I've got two bodies, one Earth sized at 0,0,0 and the other a small capsule about a megameter above the surface.  Here's the test I'm compiling with double precision enabled (same assertion fail with single precision).

```
#include "ndNewton.h"
#include <gtest/gtest.h>

TEST(Extremes, OrbitalDistances)
{
	ndWorld world;
	ndShapeInstance shapeinst(new ndShapeSphere(ndFloat32(0.5f)));

	ndMatrix matrix(ndGetIdentityMatrix());

	ndShapeInstance planetshape(new ndShapeSphere(ndFloat32(6357000)));

	ndBodyKinematic* staticbody = new ndBodyKinematic();
	staticbody->SetCollisionShape(planetshape);
	staticbody->SetMatrix(matrix);
	staticbody->SetMassMatrix(ndFloat32(5.9722e+24), planetshape);
	ndSharedPtr<ndBody> staticPtr(staticbody);
	world.AddBody(staticPtr);

	matrix.m_posit.m_y = 6357000 + 1021140;
	
	ndBodyDynamic* movingbody = new ndBodyDynamic();
	movingbody->SetNotifyCallback(new ndBodyNotify(ndBigVector(ndFloat32(0), ndFloat32(-9.81f), ndFloat32(0), ndFloat32(0))));
	movingbody->SetCollisionShape(shapeinst);
	movingbody->SetMatrix(matrix);
	movingbody->SetMassMatrix(ndFloat32(10), shapeinst);
	ndSharedPtr<ndBody> movingPtr(movingbody);
	world.AddBody(movingPtr);

	for (int i = 0; i < 480; i++)
	{
		world.Update(1.0f / 60.0f);
		world.Sync();
	}

	world.CleanUp();
}
```

Which nets me
`newton-4.00\sdk\dCollision\ndBvhNode.cpp(1260) : Assertion failed: maxGrids[0][1] < 256 * 256`

That *looks* like I'm trying to make something too big/too far away.  If this is the way it's supposed to work, I can rethink simulation model to be smaller somehow.

JulioJerez, 2023-05-25:
nothing wrong, the word size we 256 * 256 units, now can be 256 * 256 * 256

In the demo, you have to realize that a sphere can only generate one
contact, so using a sphere that large
the surface will be almost 100% flat,  but because it is defined as a
sphere it will only produce a single contact and anything on top will be on
perpetual jiter.

Fixed.
I never tried such large scale ratios, so I am curious as to how it will
work
Julio.

lperkin1, 2023-05-25:
Good to know about spheres.  This was from my orbit test for my gravity callback, so the sphere is just a place holder for Unity to render.  When I get to dealing with terrain it will have some procedural generation.

Performance wise, running 7,000 capsules in orbit with it calculating a new gravity vector every tick doesn't have a performance impact at 1000x sim speed.  Even several hundred collisions at a time has no noticeable degradation.  That's with having to update Unity transforms' position and rotation for all the objects every frame for rendering, so with just Newton I'm sure the performance would be even better.

None of the other physics engines I looked at were able to manage that at full scale.  They either didn't support doubles (Physx/Havok) in Unity, or lost over half their speed (Jolt/Bullet).  Newton's singularly impressive performance is why I decided to "bite the bullet" and make a Unity wrapper for Newton 4 even though it's meant learning some c++ (which I still suck at, as you've noticed).

Thanks for making this engine.  My project wouldn't work without it.

closed 2023-05-25 by lperkin1