MADEAPPS/newton-dynamics #316: IK Effector has no effect

lperkin1, 2023-05-31:
No more assertion failures this time :)

Next up is duplicating the basic robotic arm demo in Unity.  I got the arm bodies and joints working, but the effector wasn't being pursued in my code.  So I went back to poke at the demo and got the same results.

First I stopped the effector from generating at all.  Sure enough, the arm acts as expected, merely collapsing to the floor.
[image]

Adding back in the effector, the arm appears to point at it meaning it's having some effect
[image]

However, if I generate the effector 0.25 to the right, the arm doesn't follow
[image]

Here's the diff I have on moving the effector location, in case that messes something up.  I just changed the position of the matrix though, so I don't think it should cause a problem.
```
diff --git a/newton-4.00/applications/ndSandbox/demos/ndSimpleIndustrialRobot.cpp b/newton-4.00/applications/ndSandbox/demos/ndSimpleIndustrialRobot.cpp
index 727b16ca8..c26b6ee35 100644
--- a/newton-4.00/applications/ndSandbox/demos/ndSimpleIndustrialRobot.cpp
+++ b/newton-4.00/applications/ndSandbox/demos/ndSimpleIndustrialRobot.cpp
@@ -447,7 +447,8 @@ namespace ndSimpleRobot
                                                ndBodyDynamic* const childBody = parentBone->m_body->GetAsBodyDynamic();
                                                const ndMatrix pivotFrame(rootEntity->Find("referenceFrame")->CalculateGlobalMatrix());
-                                               const ndMatrix effectorFrame(childEntity->CalculateGlobalMatrix());
+                                               ndMatrix effectorFrame(childEntity->CalculateGlobalMatrix());
+                                               effectorFrame.m_posit.m_z += 0.25f;

                                                ndSharedPtr<ndJointBilateralConstraint> effector (new ndIk6DofEffector(effectorFrame, pivotFrame, childBody, modelNode->m_body->GetAsBodyKinematic()));
```

What am I misunderstanding?

JulioJerez, 2023-05-31:
On this one I am going to ask to wait for a weak or two.

All there demos are going over a refactoring, so that they are less
dependent of graphics in sandbox.

I am moving part of the functionality to a new library of model that can be
used as build block by an application.

There will always be some graphics simply because it needs to show the
results, but there is too much on the sandbox.

Please pin me in about a week.

lperkin1, 2023-07-02:
I got pulled into a big project at my day job, so I'm just now getting back to this.  From the recent commits it looks like you're working on the models library.  I'm guessing that means it's not ready to demo.

Is the api stable enough I can start including the classes or should I just work on other things for a while?

JulioJerez, 2023-07-02:
Is not stable enough yet. 
I am pulling all the models from both Newton Code and Sandbox demos, to place it in the ndModel library.

Thsi will be a library of effect that an application can use as a start point for any vehicle, Character or othe articulate model.
when I start I though was going to be a small task, but it require lot of supports.
so I have to stop to add that support, and I work of many things as the need arrive.
on that I got to write:
-the loader, 
-the ndModel mesh.
-The deep reinforcemnt  solvers.

so teh progress is no as far as I first estimated. But is moving alone.

I am adding lot of development on the reifirmenet learning for controlling, with the idea that an app can use to make
smart NPCs that can do more that teh normal dumb game place stuff.

anyway, you can try it out, but make sure you make a backup of your stuff.

JulioJerez, 2023-07-02:
btw: I activated the account: lperkin1

lperkin1, 2023-07-04:
Sounds good.  I was starting to build my own modeling system back when I first started getting the assertion errors.  While working through those problems, I noticed what you were building for models and scrapped what I was working on.  Your stuff is far better than anything I could make, so I'm hoping to be able to make a wrapper for it once it's stable enough.  With any luck, I'll be able to just plug in the new library to the Unity interface I've already got once it's ready.

Meanwhile, I'm thinking I'll make a pin-ball machine example to show off small scale Newton collisions, followed by a planet scale flight sim or something to work on terrain generation.  I've also got a bunch of clean up/documentation to do on what I've got for a wrapper already, so I'll work on that for a while.

I'll close this and just track the progress as new commits are made.  Thanks!

closed 2023-07-04 by lperkin1

JulioJerez, 2023-07-04:
since you show interest, I just committed the first successful reinforcement learning controller. 
It is far better that I expected. I never expected that it will take move controller forces, but is does is a very robust way.

the algorithm is of academic value, because  it can only handle discrete actions,   meaning is only knows how to push forward or backwards with a fix force or do nothing.

It is committed wit a pretrained agent.  The idea is to make sophisticated controller that can realize complex task in the environment for game play.  

it is committed wit a pre trained agent, later I will add the demo that train a new again.

lperkin1, 2023-07-05:
I'm interested alright.  I was reading the forum post about it.  I'm looking forward to messing around with the demo.  I've currently got a system for randomly generating player/character avatars of various sizes (anorexic to morbidly obese and short to tall) then generating a mesh based ragdoll off that using VHACD.

Being able to rig them with an ML controller would be a dream come true.  I was reading up on how to do it, but it was well over my head.  Glad you're making progress.