MADEAPPS/newton-dynamics #325: Please make a new tag (e.g. 4.03)

jonesmz, 2023-11-20:
This commit removed the file dCore/ndSaveLoadSytem.h : 4e567b69f25963da8e4d98e03c8fec102b7fc9c1

in git tag https://github.com/MADEAPPS/newton-dynamics/releases/tag/v4.02 for newton-dynamics/newton-4.00, this file has a syntax error.

Specifically:

```
template<class T>
class ndLoadSaveClass: public ndLoadSaveBase
{
  public:
  ndLoadSaveClass<T>(const char* const className)
  {
		RegisterLoaderClass(className, this);
	}

	virtual void* CreateClass(const ndLoadDescriptor& desc)
	{
		return new T(desc);
	}
};
```

the constructor on the 5th line of the class `ndLoadSaveClass` also has the extraneous `<T>`

This means that on many compilers, such as GCC or Clang, newton-4.00 cannot compile with tag 4.02

A new tag (e.g 4.03) would allow me to continue switching to newton 4.00.

jonesmz, 2023-11-20:
Please also include these changes in the new tag:
https://github.com/MADEAPPS/newton-dynamics/pull/322
https://github.com/MADEAPPS/newton-dynamics/pull/324

Both of them are needed to be able to build on my system

JulioJerez, 2023-11-21:
I do not understand, I can buidl an run in linux fine. I also can build on VS selecting Clang compiler
I believe I merge: 322 and 324
please sync and tell me if I still need this.
also I will make a new tag over this long weekend,  about next week. 
I have to re enable few demos that are commented out.

jonesmz, 2023-11-21:
My project only uses official releases (or git tags) of third party projects. Can't update to a specific commit.

JulioJerez, 2023-11-21:
ok, I will make teh Tag this weekend, 
If I made now, them the user that use the engine will get it and will get many demo malfunctions, and I will get many PM
Juts wait until this Saturday, and I will make that stable Tag.

jonesmz, 2023-11-21:
No need to rush, any time in the next couple of months is fine, but if you're in a good spot for it this weekend, that's great too.