MADEAPPS/newton-dynamics #327: macos-14: newton-3.14/sdk/dgCore/dgVectorSimd.h: error: unknown type name '__m128'

jonesmz, 2024-05-30:
macos-14 with the default system compiler errors out when compiling translation units which include this header.

https://github.com/TheOpenSpaceProgram/osp-magnum/actions/runs/9298107395/job/25589389932

JulioJerez, 2024-05-30:
I replaced vsprintf with vsnprintf, I do not know what to do about the simd.
somehow is missing a header of a compiler flag, which is weird afte all these years.

JulioJerez, 2024-05-30:
ok I check the header and the error log say you are build on a mac, so my guess is that you are on a newer arm mac. 

in header file E:\newton\newton-dynamics\newton-3.14\sdk\dgCore\dgTypes.h
line 142

uncomment this line. //#define DG_SCALAR_VECTOR_CLASS

you have to edit the cmake and deselect all of the simd option.  
the new macs, do not really get too much gain from explict simd anyway.  

but my best advise is that you try to go to newton 4 it is far, far better than 3. and it is  the version that active supported.