MADEAPPS/newton-dynamics #329: newtondynamics.com WIKI is not working

yuriy-chumak, 2025-01-07:
http://newtondynamics.com/wiki/index.php/FAQ is not working.

MediaWiki 1.30 internal error
MediaWiki 1.30 requires at least PHP version 5.5.9 or HHVM version 3.6.5, you are using PHP 5.4.45. 
 
[image]

JulioJerez, 2025-02-02:
Yes, I know.
unfortunately, I do no know how to fix that

gnarlyquack, 2025-02-06:
It seems you need to upgrade your version of PHP. How are you hosting your website? If it's a shared hosting provider, can't you go into your admin panel and update your software?

Alternatively, can you move the documentation to GitHub? You can add a wiki to the repository which I assume will never become outdated.

I'm trying to port a very old project to a newer version of Newton and, with the website down, there's literally no documentation about how to do this or how to use the engine.

JulioJerez, 2025-02-06:
Yes, I know, even GoDaddy have warned me that soon even the forum web hosting will go offline 
if it is not upgraded to a new PHP version.
The problem is that that is not my area of expertise. if that even happens, I see if I just move all the operation to git hup.
 
Porting to newton 4, is very easy. 
There are very simple unit test, plus some the sandbox.
take for example demo ../newton-4.00\tests\rigidBody_test.cpp
it is just this:
```
/* Rigid body must not move in the absence of forces. */
TEST(RigidBody, NoMoveWithoutForce) {
  ndWorld world;
  world.SetSubSteps(2);

  // Create a sphere at the origin. No gravity will act on it by default.
  ndVector spherePos = ndVector(0.0f, 0.0f, 0.0f, 1.0f);
  ndSharedPtr<ndBody> sphere (BuildSphere(spherePos));
  world.AddBody(sphere);

  // Sanity check: sphere must be at the origin.
  ndVector errVec = sphere->GetMatrix().m_posit - spherePos;
  ndFloat32 err = errVec.DotProduct(errVec).GetScalar();
  EXPECT_NEAR(err, 0, 1E-6);

  // Simulate one second.
  for (int i = 0; i < 60; i++) {
    world.Update(1.0f / 60.0f);
    world.Sync();
  }

  // Since no forces acted on the sphere it must not have moved.
  EXPECT_NEAR(err, 0, 1E-6);
}
```
and there are even simpler tests.
also, if you register on the forum, you can ask questions, and I can answer them.  

Julio

JulioJerez, 2025-02-06:
on this:
>It seems you need to upgrade your version of PHP

I actually try to do that, but it turns out the later PHPs are net backward compatible, and I ruing the forum. 
I had to paid godady to restore to an older backup. 
In fact, it seems GoDaddy is facing out all of the PHP forum in favor 
of World press website, but that is not a free service. it fact, it cost a lot for a project like newton.

gnarlyquack, 2025-02-07:
Unfortunately I'm not at all familiar with GoDaddy hosting, so I can't really provide any help. But if it uses cPanel, I thought that provided an interface to update the software on your server?

You're saying the forums will also go offline if the PHP version isn't updated? 😮 

I've been working on updating an [old game engine](https://github.com/FrictionalGames/HPL1Engine) to work on modern, 64-bit systems, and the last thing left is to upgrade its physics engine, which is version 1.53 of Newton. I'm actually only upgrading because the source of 1.53 isn't available, so I was planning on only updating to version 2 to try to minimize the number of backward incompatibilities, since I'm mostly interested in preserving the behavior of the engine and the game(s) it runs. As one example, `NewtonCreateBox` adds an `int shapeID` parameter, and without any API documentation available I don't know what to do to fix the function call.

yuriy-chumak, 2025-02-07:
I've extracted part of the FAQ to the github wiki using archive.org wayback machine.
https://github.com/MADEAPPS/newton-dynamics/wiki/FAQ

yuriy-chumak, 2025-02-07:
@gnarlyquack 
> I'm actually only upgrading because the source of 1.53 isn't available, so I was planning on only updating to version 2 to try to minimize the number of backward incompatibilities ... I don't know what to do to fix the function call.

I think it would be better to create a [separate issue](https://github.com/MADEAPPS/newton-dynamics/issues) with the question and accumulate all related subquestions together.

JulioJerez, 2025-02-07:
to [gnarlyquack](https://github.com/gnarlyquack)
my suggestion is to just go with 4.xx
4.xx is far more intuitive than all other versions. It is also faster and more stable.

yes there is no source code for 1.xx and even some version of 2.xx at that time I have the code in local SVS, and backups.
but after I move to google source, I relaxed and never maintain the backup until sadly lot of stuff changed (window, different machine, different media. etc) must that stuff became unusable with time. But I was not a big deal since Google code was there for even.
Them Google code when away. and we move to Github.
These days, the only history of the engine is what we find int GitHub and the fact that is open source.
Sadly, early version of 2.xx and lower where not open source, so there is not history of them.

again, just try 4.0 you will see it is far easier than any other version.  and more importantly is more robust.
for example on the case you show.

> As one example, NewtonCreateBox adds an int shapeID parameter, and without any API documentation available I don't know what to do to fix the function call.

yes, at the time there were not collision instances, therefore shapes had to contain all the information.
now, and shape is operated by an instance, and it is the instance that offers lot of flexibility.

Flareonz44, 2025-02-08:
> Yes, I know.
> unfortunately, I do no know how to fix that

Is there any other place to find tutorials or examples? I want to learn how to use it but there's not much info on other websites or YouTube.

Maybe you can upload the wiki source so it can be run locally or restore a backup.