MADEAPPS/newton-dynamics #333: Failed to build due to pch on Windows 10

orange-cpp, 2025-04-30:
```
====================[ Build | ndSandbox | Debug ]===============================
C:\Users\Vlad\AppData\Local\Programs\CLion\bin\cmake\win\x64\bin\cmake.exe --build C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\cmake-build-debug --target ndSandbox -j 6
[1/262] Building CXX object sdk\CMakeFiles\ndNewton.dir\dCollision\ndBody.cpp.obj
FAILED: sdk/CMakeFiles/ndNewton.dir/dCollision/ndBody.cpp.obj 
C:\PROGRA~1\MICROS~2\2022\COMMUN~1\VC\Tools\MSVC\1443~1.348\bin\Hostx64\x64\cl.exe  /nologo /TP -D_CRT_SECURE_NO_WARNINGS -D_D_COLLISION_DLL -D_D_COLLISION_EXPORT_DLL -D_D_CORE_DLL -D_D_CORE_EXPORT_DLL -D_D_NEWTON_DLL -D_D_NEWTON_EXPORT_DLL -D_D_SINGLE_LIBRARY -D_D_TINYXML_EXPORT_DLL -D_D_TINY_DLL -D_D_USE_AVX2_SOLVER -DndNewton_EXPORTS -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCore -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dTinyxml -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dJoints -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dIkSolver -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dParticles -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dVehicle -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dCharacter -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dExtensions\dAvx2 -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\. /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Od /fp:fast /Wall /wd4061 /wd4201 /wd4324 /wd4505 /wd4514 /wd4582 /wd4587 /wd4625 /wd4626 /wd4710 /wd4711 /wd4371 /wd4571 /wd4623 /wd4668 /wd4774 /wd4820 /wd5026 /wd5027 /wd5039 /wd5045 /wd5264 /YundCoreStdafx.h /showIncludes /Fosdk\CMakeFiles\ndNewton.dir\dCollision\ndBody.cpp.obj /Fdsdk\CMakeFiles\ndNewton.dir\ /FS -c C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBody.cpp
C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBody.cpp(22): fatal error C1083: Cannot open precompiled header file: 'ndCoreStdafx.pch': No such file or directory
[2/262] Building CXX object sdk\CMakeFiles\ndNewton.dir\dCollision\ndBodyKinematic.cpp.obj
FAILED: sdk/CMakeFiles/ndNewton.dir/dCollision/ndBodyKinematic.cpp.obj 
C:\PROGRA~1\MICROS~2\2022\COMMUN~1\VC\Tools\MSVC\1443~1.348\bin\Hostx64\x64\cl.exe  /nologo /TP -D_CRT_SECURE_NO_WARNINGS -D_D_COLLISION_DLL -D_D_COLLISION_EXPORT_DLL -D_D_CORE_DLL -D_D_CORE_EXPORT_DLL -D_D_NEWTON_DLL -D_D_NEWTON_EXPORT_DLL -D_D_SINGLE_LIBRARY -D_D_TINYXML_EXPORT_DLL -D_D_TINY_DLL -D_D_USE_AVX2_SOLVER -DndNewton_EXPORTS -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCore -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dTinyxml -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dJoints -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dIkSolver -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dParticles -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dVehicle -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dCharacter -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dExtensions\dAvx2 -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\. /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Od /fp:fast /Wall /wd4061 /wd4201 /wd4324 /wd4505 /wd4514 /wd4582 /wd4587 /wd4625 /wd4626 /wd4710 /wd4711 /wd4371 /wd4571 /wd4623 /wd4668 /wd4774 /wd4820 /wd5026 /wd5027 /wd5039 /wd5045 /wd5264 /YundCoreStdafx.h /showIncludes /Fosdk\CMakeFiles\ndNewton.dir\dCollision\ndBodyKinematic.cpp.obj /Fdsdk\CMakeFiles\ndNewton.dir\ /FS -c C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBodyKinematic.cpp
C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBodyKinematic.cpp(22): fatal error C1083: Cannot open precompiled header file: 'ndCoreStdafx.pch': No such file or directory
[3/262] Building CXX object sdk\CMakeFiles\ndNewton.dir\dCollision\ndBodyKinematicBase.cpp.obj
FAILED: sdk/CMakeFiles/ndNewton.dir/dCollision/ndBodyKinematicBase.cpp.obj 
C:\PROGRA~1\MICROS~2\2022\COMMUN~1\VC\Tools\MSVC\1443~1.348\bin\Hostx64\x64\cl.exe  /nologo /TP -D_CRT_SECURE_NO_WARNINGS -D_D_COLLISION_DLL -D_D_COLLISION_EXPORT_DLL -D_D_CORE_DLL -D_D_CORE_EXPORT_DLL -D_D_NEWTON_DLL -D_D_NEWTON_EXPORT_DLL -D_D_SINGLE_LIBRARY -D_D_TINYXML_EXPORT_DLL -D_D_TINY_DLL -D_D_USE_AVX2_SOLVER -DndNewton_EXPORTS -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCore -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dTinyxml -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dJoints -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dIkSolver -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dParticles -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dVehicle -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dCharacter -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dExtensions\dAvx2 -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\. /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Od /fp:fast /Wall /wd4061 /wd4201 /wd4324 /wd4505 /wd4514 /wd4582 /wd4587 /wd4625 /wd4626 /wd4710 /wd4711 /wd4371 /wd4571 /wd4623 /wd4668 /wd4774 /wd4820 /wd5026 /wd5027 /wd5039 /wd5045 /wd5264 /YundCoreStdafx.h /showIncludes /Fosdk\CMakeFiles\ndNewton.dir\dCollision\ndBodyKinematicBase.cpp.obj /Fdsdk\CMakeFiles\ndNewton.dir\ /FS -c C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBodyKinematicBase.cpp
C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBodyKinematicBase.cpp(22): fatal error C1083: Cannot open precompiled header file: 'ndCoreStdafx.pch': No such file or directory
[4/262] Building CXX object sdk\CMakeFiles\ndNewton.dir\dCollision\ndBodyListView.cpp.obj
FAILED: sdk/CMakeFiles/ndNewton.dir/dCollision/ndBodyListView.cpp.obj 
C:\PROGRA~1\MICROS~2\2022\COMMUN~1\VC\Tools\MSVC\1443~1.348\bin\Hostx64\x64\cl.exe  /nologo /TP -D_CRT_SECURE_NO_WARNINGS -D_D_COLLISION_DLL -D_D_COLLISION_EXPORT_DLL -D_D_CORE_DLL -D_D_CORE_EXPORT_DLL -D_D_NEWTON_DLL -D_D_NEWTON_EXPORT_DLL -D_D_SINGLE_LIBRARY -D_D_TINYXML_EXPORT_DLL -D_D_TINY_DLL -D_D_USE_AVX2_SOLVER -DndNewton_EXPORTS -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCore -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dTinyxml -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dJoints -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dIkSolver -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dParticles -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dVehicle -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dModels\dCharacter -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dExtensions\dAvx2 -IC:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\. /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Od /fp:fast /Wall /wd4061 /wd4201 /wd4324 /wd4505 /wd4514 /wd4582 /wd4587 /wd4625 /wd4626 /wd4710 /wd4711 /wd4371 /wd4571 /wd4623 /wd4668 /wd4774 /wd4820 /wd5026 /wd5027 /wd5039 /wd5045 /wd5264 /YundCoreStdafx.h /showIncludes /Fosdk\CMakeFiles\ndNewton.dir\dCollision\ndBodyListView.cpp.obj /Fdsdk\CMakeFiles\ndNewton.dir\ /FS -c C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dCollision\ndBodyListView.cpp
[... cut ...]
C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dExtensions\dAvx2\ndDynamicsUpdateAvx2.cpp(281): warning C5267: definition of implicit assignment operator for 'ndAvxFloat' is deprecated because it has a user-provided copy constructor
C:\Users\Vlad\Downloads\newton-dynamics-4.02\newton-4.00\sdk\dNewton\dExtensions\dAvx2\ndDynamicsUpdateAvx2.cpp(1145): note: implicit assignment operator for 'ndAvxFloat' first required here
ninja: build stopped: subcommand failed.
```

How to fix this to make it compile?

JulioJerez, 2025-04-30:
is this build the sandbox?

orange-cpp, 2025-04-30:
> is this build the sandbox?

yep

JulioJerez, 2025-04-30:
Did you make the project using cmake?

orange-cpp, 2025-05-01:
> Did you make the project using cmake?
> […](#)

for sure

orange-cpp, 2025-05-02:
> did you change options in cmake?

nope, just tried to build from master branch. Something with pch

JulioJerez, 2025-05-04:
I see something that might cause the problem.
for some reason that line that set the precompile header is written, as:
if ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "MSVC")

in this file. 
C:\development\newton-dynamics\newton-4.00\sdk\CMakeLists.txt

it thinks is should not have quotes 
if (${CMAKE_CXX_COMPILER_ID} STREQUAL "MSVC")

please sync and try again, see if that works.